project-generator/project_generator #51: Automagically create yaml files for a given directory

sg-, 2015-02-12:
There should be a command that creates a yaml file for the directory. There could also be standard settings ie: debug, no optimizations, etc. so all developers need to do is edit a file, not create and maintain it.

0xc0170, 2015-02-12:
How would you specify which directory? All directories within the path where was project_generator executed ?

sg-, 2015-02-15:
'''
project_generator.py -c <path with source>
'''

0xc0170, 2015-02-16:
Looks good, will look at this.

matthewelse, 2015-02-22:
Perhaps we could use argparse's subcommand feature for this sort of thing e.g.

`project_generator start_project xyz` or `project_generator init xyz`

0xc0170, 2015-02-22:
That init command is a good idea. But how would be the tree transformed to records? One big yaml file with all files in it?

sg-, 2015-02-26:
I think yaml per directory inside of /src. Kinda like a submodule would be placed.

0xc0170, 2015-03-20:
v0.4 was published. I'll close this as this functionality is there. If any new features regarding init command, any missing bits, we can create a new issue for it. THanks @sg- for reporting and helping with this

closed 2015-03-20 by 0xc0170