project-generator/project_generator #52: Eliminate tools specific settings

sg-, 2015-02-12:
https://github.com/0xc0170/project_generator/commit/f204880856437704d507950fa5a4d10ba95bf58e

Somewhere around this commit we used to have a minimal set of options for the yaml files. Then things changed and there are specific options for specific tools and I dont think this is the right approach. I think we should define what the yaml files are capable of and then let the scripts figure out the best setting when generating a project file rather than having the yaml creater also be intimately familar with the toolset. Otherwise I fear a feature like https://github.com/0xc0170/project_generator/issues/51 will never be possible.

0xc0170, 2015-02-12:
There are default settings, each tool has own definitions with predefined values. I'll look at them again, to see what is defined by default for all tools. For example https://github.com/0xc0170/project_generator/blob/master/uvision_definitions.py

The approach should be as you described, I just extended specific options to cover all options available for a tool. I prefer also keep it simple , so for many repositories, the specific tools wont be needed. We can look at them, once I land the package pull request.

0xc0170, 2015-02-12:
An example what was needed for mbed blinky example:

```
tool_specific:
    uvision:
        mcu:
            - LPC1768
        macros:
            - TOOLCHAIN_ARM_STD
            - __ASSERT_MSG
            - TOOLCHAIN_ARM
        misc:
            Cads:
                MiscControls:
                    - --debug
                    - -g
                    - --gnu
                Optim:
                    - 1
                uC99:
                    - enable
```

if mbed does not require different macros for each tool and that misc settings, there would be just mcu definition.

0xc0170, 2015-02-13:
To eliminate the mcu definition for a user, we could introduce a board definition.

For the project, there would be board. If board is not present, core and MCU needs to be defined
```
board:
    FRDM-K64F
```

This would translate to an MCU and core. We can have this predefined and users can add their boards.

0xc0170, 2015-02-13:
Created #54 for board keyword

0xc0170, 2015-02-14:
I added boards definition to #50. For a project, user defines a board in the record, and this board should be defined in the boards_definition script file, which is a translation dictionary. It translates to a value which an exporter recognizes and finds the proper board definitions (board -> cpu -> cpu definitions for the tool).

The step for using this feature, is to have support for projects boards, by time we can build a large database of boards for various tools

matthewelse, 2015-02-14:
Board definitions sound good :-) in the project I'm working on I basically
used normal records to define settings for different boards

0xc0170, 2015-02-15:
@sg- Any other ideas how to improve this? The boards are now on master.

sg-, 2015-02-15:
```
build_options:
    cpu:
        - cpu_type
    macros:
        - macros
    controls:
        - compile_controls
        - link_controls
    optimization:
        - maybe_this_is_control
```
Then the project is agnostic to a compiler and the scripts can parse the data fields and try to create files for the requested tool. So only one set of files for a project and its now easier to generate these from a directory of source and the ide is a command line option, not a bunch or settings in the files

0xc0170, 2015-02-16:
From those options , only controls and optimizations are now tool specific. We can find some common options and generalize those. Let's look at those options.

aethaniel, 2015-03-08:
hi, it seems to be where I should ask about board definitions :-p
I'm unable to find where to introduce properly these definitions.
https://github.com/0xc0170/project_generator/wiki/Contribution#board-name is not clear/complete about that and going to understand from sources is step 2.
pg mbed examples repo don't show anything also.

Any hint to point on?

Thanks

aethaniel, 2015-03-08:
OK, found board definitions inside code: project_generator/exporters/board_definitions.py

Honestly such a database should not be into code but in a separate (yaml?) file which could be superseded/augmented by project local files where pg is run. Modularity and tree file structure may be a plus also to enforce consistency once contributions on this project will grow.

Well, this are my 2cents

0xc0170, 2015-03-09:
You can create a separate issue with proposal (something like RFC) for the board definitions. Please create a new one, where we can continue the discussion.
The board name is just generalization of MCU type, which is also a python dictionary.

0xc0170, 2015-03-17:
@aethaniel I created a new issue for the board definitions, please provide feedback there.

@sg- I created couple of new issues today, some of those are implemented in the current branch. I might send a new pull request for review and testing , tomorrow

sg-, 2015-03-26:
https://github.com/mbedmicro/FlashAlgo/tree/master/records

Here is a reference of what I'd hope the record files can be as simple as. Would love to use 
```
pgen init <directory with source, headers, linker, obj, ar, etc...>
```
To get something like this or the list of directories in common form with generic tool support. tool specific name could change because I'd like to run

```
pgen -f ./records/projects.yaml -t uvision
pgen -f ./records/projects.yaml -t make
```
and have all my project files created in ../project_files/<project_name_tool>/<project_name>

Notice the generic build options in template.yaml The generator scripts could have default settings that are overridden but this is a minor point IMO because with the project files you can edit what you choose. Also the option to just use core support rather than a MCU would be great. Easy enough to select a MCU from the version of the tool you have rather than always be updating something that is already managed until CMSIS better distributes these files and settings. 

Really what is missing here (0.5) is some small config in records if it doesn't exist and then generic settings for compile rather than hard coding the tool-chain specifics

0xc0170, 2015-03-26:
That core support is there, core targets were added recerently. This is now valid target:

```
common:
  target:
    - cortex-m0
```

I''ll paste here your template.yaml:

```
common:
    source_files:
        - source
        - source/template

tool_specific:
        mcu:
            - cortex-m0
        options:
            - --gnu
            - --c99
        optimize:
            - none
        macro:
            - 
        pre_build:
            -
        post_build:
            - python ../tools/gen_algo.py %project_name%/build/%project_name%.axf
```
What is missing for v0.5, is the tool_specific standardization. It is a tough task to do for all tools we support, most probably won't make it to v0.5, but we can start adding its support there.

This requires us to choose which options to start with, come up with generic names. Create a new issue if this is not yet there, where we discuss this further. Paste there your template yaml with these options you would select for standardization.

What is this about "some small config in records if it doesn't exist" ?

matthewelse, 2015-03-26:
:-( private repositories ¯\_(ツ)_/¯

0xc0170, 2015-03-26:
@matthewelse I shared the template, that's all from there ;) read my message above

0xc0170, 2015-07-01:
I believe this one can be closed. The only misc settings currently is for gcc as it uses makefile, which we could eliminate by cmake? :)

closed 2015-07-01 by 0xc0170