project-generator/project_generator #57: Atmel Studio exporter, using UUID generation

aethaniel, 2015-03-05:
Hi Martin,

I'm investigating on bringing an exporter for Atmel Studio but unable to find any hint regarding UUID generation. I certainly miss something.
Do you know an easy way to generate UUIDs and insert the patterns in the template?

example (first lines of workspace file):
```
Microsoft Visual Studio Solution File, Format Version 11.00
# Atmel Studio Solution File, Format Version 11.00
Project("{54F91283-7BC4-4236-8FF9-10F437C3AD48}") = "samv71 - getting started", "samv71 - getting started.cproj", "{6C71DA86-4901-4A3D-A087-581D0232CB87}"
```

0xc0170, 2015-03-05:
Hi,

I haven't looked at how the atmel studio project looks like. Therefore I don't know what is the purpose of that UUID there. Can you elaborate more?

I would like to see this supported. I'll install the atmel studio and see what we can do.

aethaniel, 2015-03-05:
Hi,

I guess it is the same way than others: templates with placeholders for files and GCC/debugger options. The only difference certainly coming from Microsoft Visual Studio seems to be that each 'solution' and 'project' files have a unique ID inside.
Honestly, I did duplicate some projects in the recent past and nothing yelled at me :-D
I didn't test using UUID set to zero, anyway but for sure the UUID is not changed by Studio if present before next save.

aethaniel, 2015-03-05:
by the way, I could do the exporter without taking in consideration the UUID at first.

0xc0170, 2015-03-05:
Yes, definitely if it's possible to ignore for now, to find out more about generation of UUID

matthewelse, 2015-03-05:
I presume that the UUIDs are pretty arbitrary, but if anyone's interested I found this: https://msdn.microsoft.com/en-us/library/ms241442%28v=vs.80%29.aspx from a quick Google search.

aethaniel, 2015-03-05:
It seems to be easier than planned:
https://docs.python.org/2/library/uuid.html (from python 2.5 to infinite)

This brings more specific information about GUID as implemented by Microsoft:
http://en.wikipedia.org/wiki/Globally_unique_identifier
One day, they will follow the standards, I hope...

matthewelse, 2015-03-05:
Awesome, I clearly didn't spend enough time looking... This should be nice and easy to add

aethaniel, 2015-03-05:
let's consider this would be usable:
uuid_string_workspace = str( uuid.uuid5(uuid.RESERVED_MICROSOFT, '0xc0170') );

I'm already working on the template and patch. Let's solve potential issues step by step ;-)

0xc0170, 2015-11-19:
@aethaniel is there any progress ? :-) Tools changed quite a lot since March..

0xc0170, 2015-12-01:
Update, there is an exporter addition to mbed, which has a template defined there, that is more than a good starting point. It would be nice to have this for v0.8

aethaniel, 2015-12-03:
hi @0xc0170, sorry, it seems some of your mails didn't appear where I expected them in my mailbox.
I haven't been able to finalize this patch. AS7 files differ only a bit from AS6.2 ones.
So, I would have to revive my dev branch, if nobody else is working on this for mbed vX.Y purpose.

0xc0170, 2015-12-04:
Are they backward compatible? If we do AS7, it can be open in v6.x? Or just forward compatible?

I am not aware of anyone looking at it, but planning to do it.

aethaniel, 2015-12-04:
Afaik, it is forward compatible only in a way that AS7 propose the import of a 6.2 project. It may broke something on earlier versions.
The best would be to have two different exporters and keep safe.

Cheers

0xc0170, 2016-01-04:
I consider uuid solved (visual studio tool is generating it for instance). We should create a new tracking issue for Atmel studio integration, to track the progress

closed 2016-01-04 by 0xc0170