project-generator/project_generator #61: Set default values for templates

0xc0170, 2015-03-13:
Project like 

```
common:
  board:
  - frdm-k64f
  include_paths:
  - \gpio_demo_twrk20d72
  - \gpio_demo_twrk20d72\common
  - \gpio_demo_twrk20d72\cpu
  - \gpio_demo_twrk20d72\cpu\headers
  linker_file:
  - \gpio_demo_twrk20d72\k20x256_flash.ld
  source_files:
  - \gpio_demo_twrk20d72\main.c
  - \gpio_demo_twrk20d72\cpu\startup_k20.c
  - \gpio_demo_twrk20d72\cpu\system_k20.c
```

Should be able to export and compile , as least this one for arm gcc. Or even for uvision, although it does not accept .ld files.

matthewelse, 2015-03-13:
Why shouldn't this be able to compile?

0xc0170, 2015-03-13:
It does not at the moment, I am fixing it :-) Like optimization for gcc arm are not set to default value, and probably other missing features

matthewelse, 2015-03-13:
Oh sorry I misunderstood what you said - that makes sense

0xc0170, 2015-06-21:
There are predefined templates for v0.6, which should resolve this.Might not be optimal (I used some simple projects I had).

closed 2015-06-21 by 0xc0170