project-generator/project_generator #63: Workspace support - multiple projects in a workspace

0xc0170, 2015-03-14:
There's one project, one workspace at the moment. We can support one workspace - multiple projects.

0xc0170, 2015-04-01:
Thinking how this can be supported.

projects.yaml file:
```
projects:
    workspace_name1:
    project1:
        - 1.yaml
        - settings.yaml
    project2:
        - 2.yaml
        - settings.yaml
    workspace_name2:
        project3:
            - 3.yaml
```

tools like iar, uvision support workspaces. for makefile, we can provide one workspace makefile which includes project makefiles.

sg-, 2015-04-01:
Down with workspaces!!!! They should die!

matthewelse, 2015-04-02:
@sg-  what are your particular issues with workspaces?

sg-, 2015-04-02:
its a variable that this project solves I think. Let me turn the question around - in what case do IDE specific workspaces make sense to use versus a separate project?

matthewelse, 2015-04-02:
fair enough

aethaniel, 2015-04-03:
project1: application
project2: CSP/BSP library
project3: middleware library

-> workspace containing everything ready to be used after pgen

sg-, 2015-04-03:
Just saying that pgen can do the same thing using multiple projects and maybe introduce grouping for groups so you can build at once. From your example:
project 1: changes frequently
project 2: rarely changes
project 3: rarely changes
The rarely changes c(s)hould be archive inclusion in the application. Just a few thoughts but i'm sure there are valid uses for them, just not sure how many and if the methodology shouldn't be challenged.

Doing this within the scope of pgen (tool agnostic) would be a better way to support workspaces IMO.

0xc0170, 2015-07-04:
I got a prototype of uvision workspaces the problem is to define where to save the workspace file. I have it working like mentioned above, as grouping of projects. So export workspace_test will export both projects.

```
    workspace_test:
        k20d50m_blinky:
            - *module_mbed
            - tools/records/mbed/k20_cmsis.yaml
            - tools/records/mbed/k20_target.yaml
            - tools/records/projects/k20_blinky.yaml
        lpc1768_blinky:
            - *module_lpc1768
            - *module_mbed
            - *proj_settings
            - tools/records/projects/lpc1768_blinky.yaml
```

I'll continue with this grouping, it provides a good exercise to clean few more things :-)

With all above, its quite trivial to get workspace generaetd for like uvision. The problem is with workspace settings. How to define them? Where should it store workspace project? Each project has project_dir dictionary where it defines generated path. Workspace file should be generated somewhere also, by default to the generated path (all projects would go inside this folder), but should be also possible to overwrite it. This brings me to idea to define new workspace syntax, something like

```
    workspace_test:
       settings:
          output_dir:
              # this would overwrite projects output dir, projects would go inside this workspace dir workspace_path/project1 workspace_path/project2 worskpace_path/workspace.file
              - generated_projects/workspaces/

```

0xc0170, 2015-07-04:
To view and review, my progress is here, I'll show new ProjectWorkspace class which is slowly getting some shape. https://github.com/0xc0170/project_generator/blob/dev_workspaces/project_generator/project.py#L150 (browse on that branch).

I'll create new issue for the redesigning project/workspace

0xc0170, 2015-07-04:
So above is grouping of projects, I'll create a new issue for that. Back to workspace.

I believe the workspace is a feature on top of projects. Therefore lets propose this - in projects.yaml file, we got projects and define a new keyword workspaces.

```
workspaces:
  my_workspace:
    projects:
      - frdm_k64f_blinky
      - disco_f407vg_blinky

projects:
    frdm_k64f_blinky:
        - *module_mbed
        - tools/records/mbed/freescale_ksdk.yaml
        - tools/records/mbed/k64f_cmsis.yaml
        - tools/records/mbed/k64f_target.yaml
        - tools/records/projects/frdm_k64f_blinky.yaml
        - tools/records/mbed/frdm_k64f_target.yaml
    disco_f407vg_blinky:
        - *module_mbed
        - tools/records/mbed/disco_f407vg_cmsis.yaml
        - tools/records/mbed/disco_f407vg_target.yaml
        - tools/records/projects/disco_f407vg_blinky.yaml
```

pgen export -p for project, -w for workspace.

matthewelse, 2015-07-05:
@0xc0170 I think that 'workspaces' should just be groups of projects.

0xc0170, 2015-07-06:
The workspace is a file like a project, therefore this file needs to be exported somewhere.  What about this one:

```
projects:
    frdm_k64f_blinky:
        files:
        - *module_mbed
        - tools/records/mbed/freescale_ksdk.yaml
        - tools/records/mbed/k64f_cmsis.yaml
        - tools/records/mbed/k64f_target.yaml
        - tools/records/projects/frdm_k64f_blinky.yaml
        - tools/records/mbed/frdm_k64f_target.yaml
        settings:
            output_dir:
                name: K64F_project
                path: generated_projets/something/done

workspaces:
    workspace_test:
        projects:
            k20d50m_blinky:
            - *k20d50m_blinky
            lpc1768_blinky:
            - *lpc1768_blinky
        settings:
            output_dir:
                name: workspace_1
                path: generated_projects/workspace_1
```

output_dir can be specified for a project, hwo to specify for workspace? I dont think workspace needs a special yaml files. what if settings for a project are specified within projects.yaml file? Like above. Means for IAR, 

- generated_projects/workspace_1/workspace_test.eww
- generated_projects/workspace_1/k20d50m_blinky/ #project files here
- generated_projects/workspace_1/lpc1768_blinky/ #project files here

We just need to unify output dir for projects and workspaces.

Of course, by default , without any output dir, workspace will be generated to generated_projects/$workspace_name$ and projects belonging to this workspace to default locations if not set otherwise.

matthewelse, 2015-07-07:
How about:

```yaml
projects:
    frdm_k64f_blinky:
        files:
        - *module_mbed
        - tools/records/mbed/freescale_ksdk.yaml
        - tools/records/mbed/k64f_cmsis.yaml
        - tools/records/mbed/k64f_target.yaml
        - tools/records/projects/frdm_k64f_blinky.yaml
        - tools/records/mbed/frdm_k64f_target.yaml
        settings:
            output_dir:
                name: K64F_project
                path: generated_projets/something/done

workspaces:
    workspace_test:
        projects:
            - frdm_k64f_blinky
        settings:
            output_dir:
                name: workspace_1
                path: generated_projects/workspace_1
```

So, you initially define the project, _then_ add it to a workspace?

0xc0170, 2015-07-13:
Look at this project https://github.com/MXCHIP/MICO/tree/master/Projects/STM32F4xx, they are using workspaces, this would be nice exercise to support them. To remove projects from there and generated it with the same structure they have.

@matthewelse tinkering with the same idea about files and settings. Now that we have only projects, we can unify them somehow :-) pgen yet allows settings for a project inside yaml files, global settings in projects.

0xc0170, 2015-07-13:
Lets continue discussion here https://github.com/project-generator/project_generator/issues/192. I am closing this as dev_v0.7 branch contains workspaces support.

closed 2015-07-13 by 0xc0170