project-generator/project_generator #67: Board definitions - external location?

0xc0170, 2015-03-17:
In the #52, it was proposed by @aethaniel , to have board definitions outside of the project_generator. Or at least to be able to pass the yaml record with board definitions for a project, then use that information internally to get the right information.

```
    board_def = {
        # Freescale FRDM boards
        'frdm-k20d50m': {
            'uvision': 'MK20DX128xxx5',
            'gcc' : 'cortex-m4',
        },
        'frdm-kl25z' : {
            'iar' : 'MKL25Z128xxx4',
            'gcc' : 'cortex-m0+',
        },
        'frdm-k64f' : {
            'uvision' : 'MK64FN1M0xxx12',
            'gcc' : 'cortex-m4f',
        },

        # mbed boards
        'mbed-lpc1768' : {
            'uvision' : 'LPC1768',
            'iar' : 'LPC1768',
            'coide' : 'LPC1768',
            'gcc' : 'cortex-m3',
        },

        # ST Disco boards
        'disco_f407vg' : {
            'coide' : 'STM32F407VG',
            'gcc' : 'cortex-m4',
        },
    }
``` 

This can be expressed in the yaml file, provided externally, and the pg just gets the board. This ticket is to start a discussion how it can be implemented so it scales. The initial idea was to have a common place, but probably would be better to have a repo with all these definitions, which can be easily pulled and used.

matthewelse, 2015-03-18:
I think this would make the most sense as a repo called project_generator_definitions or something like that

aethaniel, 2015-03-18:
+1
Indeed, if it is possible it would be great and separate process from generic and global data.

0xc0170, 2015-03-19:
I believe MCU def should be also then make external, the only tool settings definitions would stay.

The one idea can be a github repository, which can be automatically pulled and parsed , to form a valid python dictionary. Or supplied a path for find this definitions locally, and use those. Let's look at options, and propose a simpleyaml file how it would look like lets say for uvision and how to pass it ot the generator

0xc0170, 2015-03-19:
An example of yaml definitions

```
board_definitions:
    frdm-k20d50m:
        uvision:
            - MK20DX128xxx5
        gcc:
            - cortex-m4
    frdm-kl25z:
        iar:
            - MKL25Z128xxx4
        gcc:
            - cortex-m0+

mcu_definitions:
    uvision:
        MK20DX128xxx5:
            TargetOption:
                Device:
                    - MK20DX128xxx5
                Vendor:
                     'Freescale Semiconductor',
                Cpu:
                    - IRAM(0x1FFFE000-0x1FFFFFFF) IRAM2(0x20000000-0x20001FFF) IROM(0x0-0x1FFFF) IROM2(0x10000000-0x10007FFF) CLOCK(12000000) CPUTYPE("Cortex-M4") ELITTLE
                FlashDriverDll:
                    - ULP2CM3(-O2510 -S0 -C0 -FO15 -FD20000000 -FC800 -FN2 -FF0MK_P128_50MHZ -FS00 -FL020000 -FF1MK_D32_50MHZ -FS110000000 -FL108000)
                DeviceId:
                    - 6206
                SFDFile:
                    - SFD\Freescale\Kinetis\MK20D5.sfr
```

matthewelse, 2015-03-19:
I think it would make more sense if the repo which contained all of the target definitions had for example:

> project_generator_devices
    |_FRDM_K64F.yaml
    |_LPC1768.yaml
    |_...

so each device (or class of devices) has its own yaml file... this could mean you could potentially have variants of devices which can be based on a different device

aethaniel, 2015-03-19:
Be careful not mix devices and boards.

I would see a kind of database organized by silicon vendor with product families and according series.

As it is usual to have common points between series a base common file could be inherited/included by a specific series or part specific definitions file.

Then we would have the board referencing the device it is based on.

This is mainly what Matthew is proposing.

On top of that, applications have their own definitions like the memory mapping, but this is next story.

My two cents.

Thibaut

0xc0170, 2015-03-19:
The problem here is mcu name. I decided to follow tools name because each tool names MCU differently (many of MCU are the same but from 10 I have seen, 2 were different, like uvision names it XYZ and IAR XYX). I agree with boards, then we have to find how to name MCU.

```
project_generator_definitions/
    boards/
        frdm-k20d50m.yaml
        mbed-lpc1768.yaml
    mcu/
        uvision/
            MK20DX128xxx5.yaml
        iar/
            MK20DX128.yaml
        coide/
           K20D50M.yaml
```

This is not ideal.. I made up the example above, but there are cases when this is true. Each tool has own naming.. So ideally would be that mcu/mk20dx128xxx5 as that one seems to be accroding their reference manual. But who decides what names is correct? :)

0xc0170, 2015-03-19:
I am taking freescale as an example , as I had some problems with choosing the right chip in tools :-)

Based on comments above, is this the one we can start with?

```
project_generator_definitions/
    boards/
        frdm-k20d50m.yaml #this will need to describe: vendor: freescale mcu: mk20dx128xxx5
        mbed-lpc1768.yaml
    mcu/
        freescale/
            mk20dx128xxx5.yaml
        nxp/
            lpc1768.yaml
```

The one who adds a board, defines MCU.. So if I know that I have own board with the same mcu, and want to add different tool, I'll add it to the same file -> mk20dx128xxx5.yaml. If my MCU is not there yet, neither the board, I'll create a new one for both. A question is, what naming convection for mcu definition yaml file should I follow? The one is written in reference manual, one in my tool I use, or one I have written on my board?

aethaniel, 2015-03-19:
Indeed, good point.

I would say the right name is the one provided by the vendor which is the one the user know from datasheet.
Then we can have an alias in device tool yaml.
Be careful that sometimes the name in tool can change, but not the one in DS.

matthewelse, 2015-03-20:
I like the idea of the aliases in the yaml file, which can include the names for different boards in different toolchains - it sort of solves the problem @0xc0170 and I were talking about.

aethaniel, 2015-03-20:
I see the repo already exists for definitions.
Can we begin to populate it a bit and then continue discussing there?

0xc0170, 2015-03-20:
That's what I am going to do today or during the weekend. I'll start porting what we have covered at the moment, boards like k64f, lpc1768 and others.

I added load command to the #70, so far no implementation, will be added once we agree on definitions structure, so I can write yaml parser for it.

matthewelse, 2015-03-21:
These are the things I can think of that need doing

- [x] Implement load command
- [ ] Decide on format for device definitions
- [ ] Add board definitions to repository
- [ ] Implement using the downloaded definitions

0xc0170, 2015-03-25:
Almost all features from the above are implemented. One more to come,it should get a new command:

```
pgen update (default - to get it from official repository, or provide git remote address to get it from)
```

0xc0170, 2015-04-02:
Resolved via #77, look at https://github.com/0xc0170/project_generator_definitions

closed 2015-04-02 by 0xc0170