project-generator/project_generator #74: Command - shorther version for project_generator

0xc0170, 2015-03-21:
I hate writing project_generator all the time, too long :neckbeard:... there are couple of options we have

```
pg
pgen
prg
prgen
```

pg is a module already in the pypi, shall we aim for another?

matthewelse, 2015-03-21:
I was starting to get fed up with this too :+1: the only other potential conflict I can see is PostgreSQL but since the kind of engineers using that are unlikely to be using project_generator too I think we'll be fine.

I think for backwards compatibility we should keep the project_generator command line name and just add another one

matthewelse, 2015-03-21:
+1 for pgen

0xc0170, 2015-03-21:
Yes, ah project_generator stays. Just need add a new to the console . 

For me +1 to pgen and prgen

matthewelse, 2015-03-21:
```python
'console_scripts': [
            "project_generator=project_generator.main:main",
            "pgen=project_generator.main:main"
        ]
```

0xc0170, 2015-04-02:
pgen is available, resolved via #77

closed 2015-04-02 by 0xc0170