project-generator/project_generator #82: Use sh or other git python modules

0xc0170, 2015-03-24:
Currently for v0.5, we are using subprocess for pulling the latest mcu definitions. As it does not provide much options, going through pipes and all the other magic to get errors, we should consider module sh or any other available options.

It should rebase to the latest master silently (or might be faster to compare local HEAD and remote HEAD). The master should be available to change.. We support currently to redirect it to the local directory. Might consider also a fork

matthewelse, 2015-03-26:
Sh looked like a really good choice, but seems like it's only supported on OSX and Linux, so alternatives are:

- Plumbumb (https://pypi.python.org/pypi/plumbum)
... I can't find any more - I'll edit this if I can find any more

0xc0170, 2015-03-26:
WE might just go ahead with subprocess as it is, just do more magic there, as I noted. If definitions have local commits -> print a warning for example

0xc0170, 2015-04-08:
I added handling if definitions HEAD is behind origin/master, then debug message is print and they are rebased. The update needs much more work, to process correctly outputs from git to provide some output what has happened.

0xc0170, 2015-10-14:
Obsolete

closed 2015-10-14 by 0xc0170