project-generator/project_generator #89: Sublime Text Support

matthewelse, 2015-04-02:
It should be fairly easy to add Sublime Text project support, using Makefile as a build tool: see here https://www.sublimetext.com/docs/2/projects.html

sg-, 2015-04-02:
Started on that but havent had time to put much effort into it. Want to do that + https://github.com/quarnster/SublimeGDB

matthewelse, 2015-04-02:
oh ok I've just started it too how much of it have you done?

0xc0170, 2015-04-02:
+1 for this feature

sg-, 2015-04-03:
https://github.com/0xc0170/project_generator/pull/93

Idea was to use make and make builder in sublimetext and the configure the gdb plugin to use pyOCD and what that becomes!! When done should be RAD!

0xc0170, 2015-04-04:
dev_sublime branch created, please send PR to this branch for sublime text support.

0xc0170, 2015-04-04:
Did some fixes, got locally a valid project, which looks like this, might help for reference

```
{
    "folders":
    [
        
        {
            "path":"../../mbed/libraries/mbed/common"
        },
        
        {
            "path":"../../mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X"
        },
        
        {
            "path":"../../mbed/libraries/mbed/targets/hal/TARGET_NXP/TARGET_LPC176X"
        },
        
        {
            "path":"../../examples/blinky"
        },
        
        {
            "path":"../../mbed/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC176X/TOOLCHAIN_GCC_ARM"
        }
        
    ],
    "build_systems":
    [
        {
            "name": "List",
            "cmd": ["ls"]
        }
    ]
}

```

0xc0170, 2015-04-04:
Looking at the template, it's json, which might be easier to produce using python json module, to have syntax correct.

0xc0170, 2015-04-06:
#96 was merged, sublime make gcc arm tool is supported, basic functionality included. I'll close this as support is there.

closed 2015-04-06 by 0xc0170