project-generator/project_generator #91: Flashing support

0xc0170, 2015-04-03:
This idea came to me yesterday. As I was testing sw previously, I was using various tools which could generate a project, build it and flash a target + run the code to get tests outputs from the console.

We have currently pgen export -b which exports and builds. I would like to investigate what we need to add to support flashing the target.

```
pgen export -p unit_test_01 --run
```

This would export, build and run the project (load, reset, run) via some configuration , which we need to design and implement.

Some information for tools:

- uVision : http://www.keil.com/support/man/docs/uv4/uv4_debug_commands.htm
- IAR : http://supp.iar.com/Support/?note=83441&from=note+26068

autopulated, 2015-04-06:
This would be awesome. It's basically what valinor is designed to do, so would move that logic out of valinor and into project_generator.

Valinor uses an external build though, so it'd be cool if project_generator would accept a path to an already-built file when doing this.

0xc0170, 2015-05-03:
#126 - initial version for flashing support, landed to dev_v0.6.

closed 2015-06-21 by 0xc0170