project-generator/project_generator #94: Improve list command

0xc0170, 2015-04-04:
We should be able to provide users a feedback, to list projects available, to list targets. to list tools.

A code example:
```
#print projects
pgen list
#print targets
pgen list targets
#print tools
pgen list tools
```

Currently, list is for list projects only.

0xc0170, 2015-06-10:
If we can get info from projects , we could expand above - if project file is defined , read that and provide a list from the projects file, otherwise provide info from pgen.

what do you think?

sarahmarshy, 2015-06-16:
This is implemented. 
#prints all projects
pgen list / pgen list projects
#prints all available tools
pgen list tools
#prints tools in specifiec project file
pgen list tools -f your_project_file.yaml
#prints all available targets
pgen list targets
#prints targets in specifiec project file
pgen list targets -f your_project_file.yaml

closed 2015-06-21 by 0xc0170

0xc0170, 2015-06-21:
@sarahmarshy pgen list projects - what should this pring ? If no -f is provided? currently there's no output.

sarahmarshy, 2015-06-22:
It shouldn't print anything. What would be logical output?