project-generator/project_generator #97: Tool from another tool and its toolchain (combined tools support)

0xc0170, 2015-04-05:
We should improve the support for tools which inherit from various other tools. As an example, sublime make gcc arm tool should pick from gcc_arm (-linker + startup), from make_gcc_arm - makefile settings, and from sublime for sublime specific tools. The last one is currently not handled.

This is the code which should be changed to resolve this issue + lines below:
```
        toolchain__specific_settings = self.tool_specific[self.TOOLCHAINS[tool]]
        tool_specific_settings = self.tool_specific[self.TOOLS[tool]]
```

0xc0170, 2015-04-05:
I added this to #97, each tool defines a toolchain and other tools which are used.

```
    TOOLS = {
        'iar_arm': {
            'toolchain' : 'iar',
            'toolnames' : ['iar_arm'],
        },
        'uvision': {
            'toolchain' : 'uvision',
            'toolnames' : ['uvision'],
        },
        'coide': {
            'toolchain' : 'gcc_arm',
            'toolnames' : ['coide'],
        },
        'make_gcc_arm': {
            'toolchain' : 'gcc_arm',
            'toolnames' : ['make_gcc_arm'],
        },
        'eclipse_make_gcc_arm': {
            'toolchain' : 'gcc_arm',
            'toolnames' : ['eclipse_make_gcc_arm', 'make_gcc_arm'],
        },
        'sublime_make_gcc_arm' : {
            'toolchain' : 'gcc_arm',
            'toolnames' : ['sublime_make_gcc_arm', 'make_gcc_arm', 'sublime'],
        },
        'sublime' : {
            'toolchain' : None,
            'toolnames' : ['sublime'],
        },
    }
```
toolnames defined other tools used, so for sublime_make_gcc_arm, sublime, make_gcc_arm and gcc_arm will be picked to form a project. This can be scattered to each own tool

0xc0170, 2015-04-06:
Class ToolsSupported which has this list, the above plus some other data, there were in tool script file (export, build) and in the project were tools "repeated".

```
    # Tools dictionary, defines toolchain and all tools used
    TOOLS = {
        'iar_arm': {
            'toolchain' : 'iar',
            'toolnames' : ['iar_arm'],
            'exporter' : IAREWARMExporter,
            'builder' : IARBuilder,
        },
        'uvision': {
            'toolchain' : 'uvision',
            'toolnames' : ['uvision'],
            'exporter' : UvisionExporter,
            'builder' : UvisionBuilder,
        },
        'coide': {
            'toolchain' : 'gcc_arm',
            'toolnames' : ['coide'],
            'exporter' : CoideExporter,
            'builder' : None,
        },
        'make_gcc_arm': {
            'toolchain' : 'gcc_arm',
            'toolnames' : ['make_gcc_arm'],
            'exporter' : MakefileGccArmExporter,
            'builder' : MakefileGccArmBuilder,
        },
        'eclipse_make_gcc_arm': {
            'toolchain' : 'gcc_arm',
            'toolnames' : ['eclipse_make_gcc_arm', 'make_gcc_arm'],
            'exporter' : EclipseGnuARMExporter,
            'builder' : None,
        },
        'sublime_make_gcc_arm' : {
            'toolchain' : 'gcc_arm',
            'toolnames' : ['sublime_make_gcc_arm', 'make_gcc_arm', 'sublime'],
            'exporter' : SublimeTextMakeGccARMExporter,
            'builder' : MakefileGccArmBuilder,
        },
        'sublime' : {
            'toolchain' : None,
            'toolnames' : ['sublime'],
            'exporter' : None,
            'builder' : None,
        },
        'gdb' : {
            'toolchain' : None,
            'toolnames' : ['gdb'],
            'exporter' : GDBExporter,
            'builder' : None,
        },
        'arm_none_eabi_gdb' : {
            'toolchain' : None,
            'toolnames' : ['gdb'],
            'exporter' : ARMNoneEABIGDBExporter,
            'builder' : None,
        },
    }
```

matthewelse, 2015-04-06:
Another way I thought about doing this was something like this:

```python
class SublimeMakeGCCARMExporter(MakeGCCARMExporter, SublimeExporter):
    # specific stuff goes here
```
if this needs to be stored in an array or dictionary, then you could use a decorator like so:
```python
@register_exporter('sublime_make_gcc_arm')
class SublimeMakeGCCARMExporter(MakeGCCARMExporter, SublimeExporter):
     # ...
```

0xc0170, 2015-04-06:
It should be as a dic, as there will be other tools which define own projects data. That decorator might be. I'll merge the above solution, we can change it afterwards.

0xc0170, 2015-04-07:
@matthewelse I am keeping this open as we can improve this inheritance (=how to define it properly). I merged one round of cleaning

closed 2015-04-21 by 0xc0170