project-generator/project_generator #98: Visual studio support

0xc0170, 2015-04-06:
A good start would be to support visual studio makefile based project.

georgeedwards, 2015-11-15:
I agree - this would be very helpful. Is any work being done on this and is there anything I can do to help?

0xc0170, 2015-11-16:
@georgeedwards 

I haven't used visual studio for years, thus need to download and look how are project defined (their project syntax). 

We could add this for new upcoming v0.8, there's a branch. If you need any help, or docs are not clear, let me know, I'll fix it. There's 2 sections in our wiki, Tools and Templates - https://github.com/project-generator/project_generator/wiki/Tools. The starting point in the code - https://github.com/project-generator/project_generator/blob/dev_v0.8/project_generator/tools/tool.py

If you can share information what's your setup.

georgeedwards, 2015-11-16:
@0xc0170 

It would be great to see in in v0.8. I use Visual Studio 2015 Enterprise, which organises C++ projects into blank or with Makefile. Not sure if there is a configuration better suited to your tools? I will assume Makefile, as that is what the GNU projects use. 

VS organises in Solutions, with projects inside. The Project then organises files into:

*External Dependencies
*Header Files
*References
*Resource Files
*Source Files

However, this is generated by the VS GUI, the file paths are actually structured like this. In your roject file, there is the Solution file (.sln) and a (.sdf) file. Then all the content is in a file (Project_Name). Which contains Project_Name.vcxproj and Project_Name.vcxproj.filters and readme.txt. Then any content you create, i.e. main.cpp etc. 

I'll take a look at the docs and see what I can do. Thanks for your help.

0xc0170, 2015-11-16:
You use it as an editor only? I could not find free available plugin for general embedded development. visualmicro is for arduino, visualgdb is not free.  Did I miss anything? I would like to have a similar concept there as for eclipse where is Makefile Project. We would generate a makefile which contains all data to build an app, and sources are in the project view to be able to edit/debug. This should be feasible, look at how eclipse tool  is implemented. Use MakefileGCCARM class, and create a template (the xml format seems to be simple, you can use jinja2 to define a template). 

An example of setup can be found for example for platformio: https://github.com/platformio/platformio/blob/develop/docs/ide/visualstudio.rst

Share the progress, would be nice to see PR for dev_v0.8. One warning, I am trying to fix few internal things like fix_paths methods in tools at the moment

0xc0170, 2015-11-21:
@georgeedwards Update from me. There was gdb plugin for VS released yesterday. I was able to run my makefile k64f demo there with pyOCD. I had to overcome some issues like make not recognized or with paths, I hope to fix this with progen.

I started created a template for visual studio (for upcoming 0.8 version). I am considering adding few tools:
- vs_make_gcc_arm_pyocd (Visual studio with Makefile for GCC ARM , with debugger set up for pyOCD)
- vs_make_gcc_arm (no debugger)
I would like to show how we can support various gdb, openOCD or JlinkGDB which can come later as small additions.

I can push my progress to a branch and share it here. Any help appreciated !

0xc0170, 2015-11-21:
I pushed my progress to https://github.com/0xc0170/project_generator/tree/dev_visual_studio , there's visual_studio.py. I'll write there step by step how I am progressing, will become a new wiki page "how to create a new tool".

0xc0170, 2015-11-21:
Look at #312 , partially working, need to fix debugger settings (firstly pyOCD). To test, I am using lpc1768_blinky from this repo (https://github.com/project-generator/project_generator_mbed_examples). Command : ``` progen -vvv generate -f projects.yaml -p lpc1768_blinky -t visual_studio_make_gcc_arm ```

0xc0170, 2015-12-12:
@georgeedwards I releasev 0.8, it includes visual studio support. There are some limitations which I described in the PR , I believe. Briefly - I could not figure out how to edit paths in visual studio, I always get make not recognized however make is owrking outiside VS. Plus the current plugin does not support gdb commands - I could only connect to gdb server and then debug (not to load the executable, halt/reset mcu, etc). 

I am closing this as VS is there, and will try to improve the implementation in the upcoming releases.

closed 2015-12-12 by 0xc0170