project-generator/project_generator #101: Common toolchain settings

sg-, 2015-04-07:
I currently have:
```
projects:
    template:
        - records/projects/template.yaml
        - records/tools/uvision4.yaml
        - records/tools/gcc_make.yaml
    nrf51xxx:
        - records/projects/nordic/nrf51xxx.yaml
        - records/tools/uvision4.yaml
        - records/tools/gcc_make.yaml
```
and you can see now much copy and paste would be needed here. I'd like to do something like:
```
projects:
    - records/tools/uvision4.yaml
    - records/tools/gcc_make.yaml
    template:
        - records/projects/template.yaml
    nrf51xxx:
        - records/projects/nordic/nrf51xxx.yaml
```
but doesn't currently seem possible. Maybe I missed a label that does this. It would then make sense to reorganize /generated_projects to be:
```
\---generated_projects
    +---gcc
    |   +---project_1
    |   +---project_2
    |   +---project_3
    |   \---project_n
    +---sublime
    |   +---project_1
    |   +---project_2
    |   +---project_3
    |   \---project_n
    \---uvision
        +---project_1
        +---project_2
        +---project_3
        \---project_n
```

0xc0170, 2015-04-07:
there's settings section in the projects.yaml which is valid for all projects within a repo. A user can define there paths to tools which are always valid for all projects. If the projects are not changing their settings, they can be defined there (overruled by separate proejct settings anyway).

An example:
```
settings:
    iar:
        paths:
            - path to IAR
        settings:
            - yaml file with project settings
```

sg-, 2015-04-07:
I'd like to see these implemented in yaml files such that the following is valid and target settings can be in a project yaml file. :dancers: 
```
projects:
    tool_settings:
        - records/tools/uvision4.yaml
        - records/tools/gcc_make.yaml
```

0xc0170, 2015-04-11:
This is what I came up with

```
settings:
    projects:
        - tools/records/mbed/proj_set.yaml
    tools:
        iar:
            path:
                - path
```

Settings per projects and tools,

0xc0170, 2015-04-12:
Settings is confusing as proposed above.. It would be ideal to do : 

```
projects:
    - records/tools/uvision4.yaml
    - records/tools/gcc_make.yaml
    template:
        - records/projects/template.yaml
    nrf51xxx:
        - records/projects/nordic/nrf51xxx.yaml
```
But it's not valid YAML syntax. We can introduce a keyword projects_common, however I am not inclined to this, neither in my proposal above.

This code below does not seem to be the correct way neither, as all keys within projects are valid projects to be exported, not merging tool_settings with the other projects. Firstly, I was thinking about common key word to be included in the all projects, it just complicates things. 
```
projects:
    tool_settings:
        - records/tools/uvision4.yaml
        - records/tools/gcc_make.yaml
```

0xc0170, 2015-04-12:
I created a new issue for reorganize output dir structure.

0xc0170, 2015-04-13:
So far, the simplest way to do this, with the new feature I would like to implement - pgen keywords, we could do this - include projects (a project is a keyword which pgen can understand).

```
projects:
    template:
        - records/tools/uvision4.yaml
        - records/tools/gcc_make.yaml
        - records/projects/template.yaml
    nrf51xxx:
        - $template$
        - records/tools/uvision4.yaml
        - records/tools/gcc_make.yaml
        - records/projects/nordic/nrf51xxx.yaml
```

It's confusing that template is not real project, it's just a group of yaml files which are common for some projects. What about adding modules to main record file, to group yaml files. This will allow some projects to group common "modules" (yaml files) to groups and then jsut include them.

```
modules:
    template:
        - records/tools/uvision4.yaml
        - records/tools/gcc_make.yaml
        - records/projects/template.yaml

projects:
    nrf51xxx:
        - $template$
        - records/tools/uvision4.yaml
        - records/tools/gcc_make.yaml
        - records/projects/nordic/nrf51xxx.yaml
```

0xc0170, 2015-04-13:
```
# in projects.yaml
projects:
    k20_blinky:
        - k20_blinky.yaml

```

```
# k20_blinky.yaml
common:
    target:
        - frdm-k20
    include_yaml:
        - records/common/uvision4_settings.yaml
        - records/common/gcc_settings.yaml
        - records/common/mbed_common.yaml
```

The inclusion of other yaml file would be valid only for the project yaml file. The priority would be:
yaml includes as they are defined, project yaml files (one or many). Means that the top one has the lowest priority.

0xc0170, 2015-04-15:
Thanks to @matthewelse for referencing YAML features, I found there anchors and references, which provides exactly what we need.

```
modules:
    mbed: &module_mbed
        - tools/records/mbed/common.yaml
    lpc1768: &module_lpc1768
        - tools/records/mbed/lpc1768_cmsis.yaml
        - tools/records/mbed/lpc1768_target.yaml
    proj_settings: &proj_settings
        - tools/records/mbed/proj_set.yaml

projects:
    k20d50m_blinky:
        - tools/records/mbed/k20_cmsis.yaml
        - tools/records/mbed/k20_target.yaml
        - tools/records/projects/k20_blinky.yaml
    lpc1768_blinky:
        - tools/records/projects/lpc1768_blinky.yaml
        - *module_lpc1768
        - *module_mbed
        - *proj_settings
```

Create &name anchor, then reference it via *name. I'll send PR supporting this (just one change is required).

closed 2015-04-15 by 0xc0170

0xc0170, 2015-04-15:
@sg- this got automatically closed. As those common settings are now on master

PR #113 (2015-04-15): workspace - support YAML references for projects
This should fix #101.

I'll wait for tests to be fixed, then release v0.5.7