project-generator/project_generator #118: Organization project-generator

0xc0170, 2015-04-18:
Shall we move all repositories to the organization I created for pgen? This was one of the available options, so it's live: https://github.com/project-generator

There are already few of repostiories,and it will grow :smile_cat:
- pgen
- pgen definitions
- pgen tests
- pgen mbed example

matthewelse, 2015-04-18:
If you keep all of the repositories open as forks on your github then there's backwards compatibility

0xc0170, 2015-04-18:
True that ! Will do the next week

0xc0170, 2015-04-19:
@matthewelse Can you move tests there? Ill update travis to poin to a new loc, will fork it to fix the latest changes in the PR which is open

Pgen org needs a logo :)

0xc0170, 2015-04-21:
definitions and mbed example moved, pgen will follow soon with updates to all references.

0xc0170, 2015-04-21:
All repo were moved, project-generator donE !

closed 2015-04-21 by 0xc0170