project-generator/project_generator #119: pgen v0.6

0xc0170, 2015-04-19:
There's a milestone for v0.6 which is similar to what I came with yesterday while being offline.

What we should aim for v0.6?
    - build any project
    - copy sources to exported dir - pgen use by IDE? :)
    - fix definitions from local sources
    - sublime - fix the template to be more generic
    - testing ! please write some :-)
    - debuggers support in records and their definitions for tools - this might be crucial for flashing

sg-, 2015-04-20:
- generic default settings for all tools
- create project.yaml from any given directory
- add exporter support for Qt Creator

matthewelse, 2015-04-20:
* perhaps add support for Visual Studio?

matthewelse, 2015-04-20:
* Add shell completion (like ARMmbed/yotta#222) using https://github.com/kislyuk/argcomplete

sg-, 2015-04-21:
I wanted to add some extra weight to:
- build any project, build all projects for a toolchain

0xc0170, 2015-04-24:
I created dev_v0.6 branch for development. On master - only bugfixes for v0.5, anything new, please send it to dev_v0.6 version. Thanks, and lets start developing new minor version.

0xc0170, 2015-04-25:
Guys, thanks for your inputs ! Can you check if all of yours have an issue created? There's milestone v0.6, which we can assign to all these and track the progress. Thanks

0xc0170, 2015-05-03:
I created issues for all proposal here, please check it here : https://github.com/project-generator/project_generator/milestones/v0.6

0xc0170, 2015-06-21:
Looks like all issues created for v0.6 were resolved. I moved some to long term as there's enough feature/breaking changes on dev_v0.6 branch. Period of testing (yes unit test still are not implemented, lets remind us once in a while) and bugfixing !

0xc0170, 2015-06-25:
v0.6 was released. I'll create a new issue for v0.7 and also new milestone, to define goals.

Thanks guys for your inputs !

closed 2015-06-25 by 0xc0170