project-generator/project_generator #149: Tools project parser

0xc0170, 2015-05-31:
What I am considering is to create a parser for tools projects (uvision, iar, coide, etc). Note: a parser would be for those templates which have predefined syntax, not a user makefile for example

Usage: A user would open ide of their choice, create a new project (or use already working project), place it somewhere within a repo, add its path to project or projects (global or per project). This project would be parsed to get settings (pgen knows all it should to get proper data and use it) and exported. As a start, it would be about settings -> paths and files would be ignored.

In the projects.yaml file:
```
settings:
    tools:
        iar:
            template:
                name:
                    - name of the project
                path:
                    - pathtotemplatefile #pgen would seek there for files it requries for exporting
```
That template file would be used for all IAR projects within a repository.

What do you think? Would this be useful and how we can improve to eliminate those tools_specific options?

sg-, 2015-06-01:
I think this would be useful for people adding support for new chips but if the targets were well defined the devs should only have to run a cmd line to create the proper files:
```
pgen create -t iar -m mcu123
``` 
for iar only or 
```
pgen create -m mcu123
```
for all supported ide

0xc0170, 2015-06-01:
Ah yes, another use would be for mcu, fair point!
What would be the use ? For the project settings, template to use. But for the new chip addition, create a mcu yaml file ,which then can be sent to definitions as a new addition? Something to think about.

0xc0170, 2015-06-01:
At the moment, got some basic template parsing for uvision, my branch is https://github.com/0xc0170/project_generator/tree/dev_project_parser

As I am testing it with mbed example repo, I defined this in the projects.yaml:
```
settings:
    tools:
        uvision:
            template:
                name:
                    - template
                path:
                    - .
```
Created template.uvproj, and settings from this proejct should be used. Needs more testing, further parsing and refactoring of templates :-)

0xc0170, 2015-06-21:
#156 - merged, this is now resolved.

To specify a template for projects:
```
settings:
    tools:
        iar:
            template:
              - path_to_template.ewp
```

for a project:

```
tool_specific:
  iar:
    template:
      - path_to_template.ewp
```

closed 2015-06-21 by 0xc0170