project-generator/project_generator #150: Templates - iar, uvision and coide needs some refactoring

0xc0170, 2015-06-02:
As those xml are not that easy to go through the templates are not that correct, the dictionaries do not reflect the syntax used there. 

For example IAR has 9 main attributes - General, ICCARM, AARM , ... . We should refactor those templates. I came across this while writing that parser and can finally see nicely the xml in the python dictionary.

A note - we will need to dig into wiki after all recent changes for v0.6 :)

sg-, 2015-06-02:
Agree - Need to either spend time really defining the templates or only one breaking change to them per minor release :+1:

0xc0170, 2015-06-02:
Once I parse all of them, I'll get better view of the xml structure for each of them, will create a tickets  with the proposed misc structure which will be turn into the documentation.

0xc0170, 2015-06-21:
Solved via #156.

Refactors. uvision and iar are not using jinja2 templates, will be removed later. coide stays with jinja2 for now, was not able to generate valid project. They dont report much errors, just error so hard for a user to find out what is wrong with a project. Not to mention their 2.0 beta version adds new options.

closed 2015-06-21 by 0xc0170