project-generator/project_generator #155: Obj or lib sources

0xc0170, 2015-06-10:
I have consolidated sources recently for v0.6, and now thinking what we can do with

- source_files_obj
- source_files_lib

We probably can't make all into sources, as some libs might be without extension (although do we support  that?). I propose 2 solutions

1. All in sources - object files, libraries, source files (requires to fill proper extensions for us to group internally)
2. source_files_obj rename to object_files, source_files_lib to static_libraries
3. 1 and 2 combined - to resolve issue in the 1st one - using the second if needed. This would mean to keep both formats possible, which might be confusing.

sg-, 2015-06-11:
+1 I think all should be in sources

sarahmarshy, 2015-06-15:
My latest commit puts all in sources

sarahmarshy, 2015-06-16:
Removed "group_name" section in yaml

PR #159 (2015-06-19): Allows group names for .obj and .lib. Parses directories for possible…
… relative include paths in header files. This fixes #155, fixes #146, fixes #140
commit 62e9d314 (2015-06-19): Allows group names for .obj and .lib. Parses directories for possible relative include paths in header files. This fixes #155, fixes #146, fixes #140
closed 2015-06-21 by 0xc0170