project-generator/project_generator #158: Generate_dict_for_tool function in projects.py

sarahmarshy, 2015-06-16:
I would like to eliminate this functon, as it can be done when we are parsing the projects file.

matthewelse, 2015-06-17:
@sarahmarshy this is how it was done before I refactored the code into a Project class. I think that, in fact a better way of dealing with this issue is by using instances of the project class in the template exporter code instead of passing round dictionaries.

Since some of the data has to be formatted (e.g. putting all of the c++ source files into one list), this can be implemented as a method in the Project class, which can be called from the Exporter class.

0xc0170, 2015-07-02:
@sarahmarshy Can you look at this issue ? We should reorganize things a bit. Tools are done, now look at this how project is created and data passed around

0xc0170, 2015-07-09:
@sarahmarshy Now we have customize_project_for_tool(), any more plans for this issue?

closed 2015-07-15 by 0xc0170