project-generator/project_generator #160: Replace jinja2 templates by project templates

0xc0170, 2015-06-17:
Relates to #149, as pgen will get project parsing support (a user provides own project aka template), most of jinja2 templates should be replaced by exporter dictionaries.

Each exporter will have a dictinary which is 1:1 mapping to the project files. This dictionary will be adjusted to the user's input (template, yaml files), then exported.

This should eliminate jinja2 templates for iar, uvision and coide. This should simplify adding new exporters and unify them a bit, as now misc options are not 1:1 mapping to the tools projects format.

I'll provide more details once I make it working for IAR.

sg-, 2015-06-17:
no more v0.6!!! Need a release :D

0xc0170, 2015-06-18:
@sg- Doing what I can to complete v0.6 ! I find this as the most important change which should happen as soon as possible :-) We need contribution to other IDE, this issue is important. 

I am away for the next week, hope to complete it this week :smile:

0xc0170, 2015-06-21:
OK, looks like coide for now is using jinja2 template but project parser is there in place, jinja2 was modified to enable 1:1 mapping. Was merged, so this should be fixed for v0.6

closed 2015-06-21 by 0xc0170