project-generator/project_generator #161: Parse a project file to create a project yaml file

0xc0170, 2015-06-18:
Just a thought, but with the latest changes for v0.6, the exporter knows the structure of the tool projects. Therefore it can be possible to supply projects files, pgen could create a valid yaml project file. Means otherway around - from project files to pgen YAML. Currently we do from pgen YAML to  project files.

As an example - got uvision project in my repository which compiles, I would like to start using pgen to generate this project or add IAR support. A command like:

```
pgen create project -dir uvision_project
```

Pgen would get all common data from the project, created project yaml file.

matthewelse, 2015-06-18:
I think that the following syntax would possibly make more sense:
```sh
pgen import abcdef.uvproj
```

0xc0170, 2015-06-18:
We need to unify create/import. So far, we got creating yaml files from a repository, this will add creating yaml file from a project, then we will have create yaml target definition from a project file. All these 3 are about creating yaml files. Thus I stayed with create

We have only pgen init - to create yaml file from a repository. I am proposing to unify all these 3.

```
pgen create target
pgen create project (-dir or -file)
```

With import, we could have

```
pgen import project -f projectfile #this one might need tool selection as many eclipse based have .cproject file which I am not certain if we can distinguish the right tool, something to look at
pgen import target projectfile # create yaml target file - for adding a new targets.
pgen import project -dir #if its a directory, we know we are creating project yaml file
```

Ideas?

0xc0170, 2015-06-18:
In favor of don't do breaking changes that often, we could unify this create/import/init for v0.6 :-)

sarahmarshy, 2015-06-22:
So, it should be one command then? "Import" with functionality of current init and added proposed changes.

sarahmarshy, 2015-06-22:
Can you expand on the "pgen import target projectfile"? Also, proposing to change that to "pgen import target -f projectfile" to match the format of others.

I am starting up development on this.

0xc0170, 2015-06-22:
pgen import target -template (or some argument or no argument as it can be only a project file plus tool so we can get all required files - assuming that a target might be stored in multiple files). This would get mcu info which is stored in the pgen definitions (look at mcu folder), that's the format of yaml file it would create.

0xc0170, 2020-01-04:
We do import targets via `progendef` module. To get a project from the tool though, another nice feature but writing yaml file should be quick and simple for a project, I don't believe we need to get that smart to start parsing IDE projects to create yaml. Use sources what you have to create yaml project files. I'll close as won't fix.

closed 2020-01-04 by 0xc0170