project-generator/project_generator #164: Documentation - from wiki to gh pages?

0xc0170, 2015-06-22:
Github wiki is not ideal, therefore looking for options to make it more collaboration friendly. I'll update wiki to v0.6 version, as it will be the last one (v0.7 should be shipped with new structured docs)

First proposal - gh-pages, to have docs folder in this repo.

matthewelse, 2015-06-22:
Yeah that sounds like a good plan - another possibility is readthedocs

matthewelse, 2015-06-24:
@0xc0170 I think I'll have a look at moving the documentation to readthedocs

matthewelse, 2015-06-24:
actually we could just use gh-pages

matthewelse, 2015-06-28:
I'm going to work on documentation on my branch for the time being, otherwise I'll end up creating more and more prs.

0xc0170, 2015-06-28:
Ok, I'll keep an eye on your branch, and can do some help there ;)

0xc0170, 2015-07-09:
@matthewelse Once you return, can we look at docs? My focus for 0.7 is to deliver tests to cover basic functionality, documentation up to date in doc folder and some insight redesign which we started. Let me know what's the status - share the work with us, we can help

0xc0170, 2015-07-11:
I am starting writing new docs, will keep you posted.

0xc0170, 2020-01-04:
Let's keep wiki up to date, should be sufficient.

closed 2020-01-04 by 0xc0170