project-generator/project_generator #166: pgen init rename

0xc0170, 2015-06-24:
As in #161, we should unify init command. @matthewelse proposed import. We started having that discussion there, lets redirect it here.

We currently have pgen init which creates yaml project file. At least, it will need to have additional functionality covered - create mcu definition file, create tool specific yaml file,  create project yaml file from an existing project). 

I assigned this to v0.6, as we should at least unify the command name - rename init to import and provide an option which we currently support - importing a repository to pgen yaml file. This allows us to add additional arguments.

@sarahmarsh Any progress with this renaming?

sg-, 2015-06-24:
please release 0.6 - no more breaking changes :D Lets break more in 0.7

matthewelse, 2015-06-24:
I agree with @sg-

sarahmarshy, 2015-06-25:
@0xc0170 small progress. Have a working command. I've created a new method in project.py "create_yaml_from_file," that retrieves information from a given project file. I am still working out how to parse this information, but, it is there.

sarahmarshy, 2015-06-25:
What is the advantage of going from tool to yaml file when the means of going from directory to yaml file already exists?

matthewelse, 2015-06-25:
@sarahmarshy I would imagine that the main advantage is other information
other than files - build settings, defines etc

0xc0170, 2015-06-30:
Yes, the other information as @matthewelse pointed out. But thats for another issue to discuss. This one is for renaming init.

closed 2015-07-26 by 0xc0170