project-generator/project_generator #177: Tools should have own class (exporter, builder, flasher)

0xc0170, 2015-06-29:
tools are scattered in different layers. There should be tool classes like IAR, uVision, etc which define API for exporting, building and flashing.

I started doing this on my branch https://github.com/0xc0170/project_generator/tree/dev_iar_tool for IAR, to see how it will look and will do for uvision and see how much tools have in common.

cc @sarahmarshy @matthewelse @sg-

matthewelse, 2015-06-30:
+1 this definitely makes sense

0xc0170, 2015-06-30:
Guys, I made some good progress. IAR file contains all within, definitions, IARWorkbench class defines export, build and flash methods, plus all additional sugar required.

0xc0170, 2015-06-30:
IAR and uvision done so far

0xc0170, 2015-07-01:
All tools were moved to tools/ dir.

closed 2015-07-01 by 0xc0170