project-generator/project_generator #186: Workspace - init - list of records - unordered?

0xc0170, 2015-07-04:
@sarahmarshy what's the intention of using set here? https://github.com/project-generator/project_generator/blob/master/project_generator/workspace.py#L47

The order in project records is important as data can be overwritten. The last one has the highest importance. This does not maintain order therefore might have unexpected results.

An example:

```
        k20d50m_blinky:
            - *module_mbed
            - tools/records/mbed/k20_cmsis.yaml
            - tools/records/mbed/k20_target.yaml
            - tools/records/projects/k20_blinky.yaml
```

the list of records for me is - cmsis, blinky, target, common.

matthewelse, 2015-07-06:
I think that the idea behind this was to remove items which are repeated, and this is affected by #187, where the set(...) code is moved to util. We can probably look at finding another de-duplication method that maintains order. I'll add this to the workspaces branch, since it's modified there.

0xc0170, 2015-07-07:
As I recall, this one is resolved on v0.7 branch

closed 2015-07-07 by 0xc0170