project-generator/project_generator #188: Grouping of projects

0xc0170, 2015-07-04:
On my way  to support workspaces, I implemented a new feature, which we might want to add. Grouping of projects.

Works as, that a user creates a group of projects.

```
projects:
  bootloader:
    freescale_bootloader:
      - files
    nxp_bootloader:
      - files
  flasher:
    freescale_flasher:
      - files
  project_no_group:
    - files
```

pgen export -p bootloader or should be valid also pgen export -p nxp_bootloader

matthewelse, 2015-07-04:
Why is the -p flag required, couldn't it just be `pgen export bootloader`?

0xc0170, 2015-07-04:
That's for another issue, currently it's export -p as I recall

0xc0170, 2015-07-13:
Present on dev_0.7, groups are a workspace. @matthewelse we should create an issue for -p not needed and implement it

closed 2015-07-13 by 0xc0170