project-generator/project_generator #192: Unify settings

0xc0170, 2015-07-07:
Just a thought here captured.

This came to my mind when was about to add workspaces. currently, we support in common various settings which are valid for project only. Lets illustrate

common:
  target:
  - frdm-k64f
  export_dir:
    name:
    - freedom_project
    path:
    - path/some/folder
  sources:
  - main.cpp
  macros:
  - DEMO_MACRO
The above mixing in common multiple keywords, sources, includes, and project specific settings like target, output, export dir and others.

I proposed this for workspaces:

workspaces:
    workspace_1:
        projects:
            k20d50m_blinky:
            - *k20d50m_blinky
            lpc1768_blinky:
            - *lpc1768_blinky
        settings:
            export_dir: 
                - 'generated_projects\$workspace$\$tool$_$project_name$_$target$'
Shall we unify settings? We said that everything is in YAML file, so for projet,there would be project_settings. The first example would be:

common:
  sources:
  - main.cpp
  macros:
  - DEMO_MACRO

project_settings:
  target:
  - frdm-k64f
  export_dir:
    name:
    - freedom_project
    path:
    - path/some/folder
For workspace, workspace_settings, which could be included to workspace as:

workspaces:
    workspace_1:
        projects:
            k20d50m_blinky:
            - *k20d50m_blinky
            lpc1768_blinky:
            - *lpc1768_blinky
        settings:
            - workspace_1.yaml # here workspace_settings defined, export dir at the moment would be there

sarahmarshy, 2015-07-07:
workspaces:
        workspace_1:
            projects:
                k20d50m_blinky:
                - *k20d50m_blinky
                lpc1768_blinky:
                - *lpc1768_blinky
            settings:
                export_dir: 
                    - generated_projects\$workspace$\$tool$_$project_name$_$target$

Above is supported in my branch dev_v0.7

    common:
       stuff
    projects:
        stuff
    settings:
        stuff

In my branch, the settings defined at that point above would be for all projects.

I think that if a user really wants custom export directory seperate from the one defined in general setttings, they should make a workspace. If they are separating one project from the rest, that is a new workspace outside the default. 

@matthewelse what do you think?

0xc0170, 2015-07-09:
@sarahmarshy Need your update how it is now implemented, I am confused from what I am seeing and getting as result. I suggest writing tests for export_dir for project and for workspace with that string template.

We should cover various combinations, workspace export path/no project paths defined, default settings for all export paths, or some of those defined by a user.

- default: default_dir/workspace_name/all projects dir here (this will be probably most used option)
- various combinations - workspace_export_dir/workspace_file and projects are generated where defined (either default - inside the workspace, or placed to defined users path) . An example - workspace in default_dir/workspace_name and workspace projects - in examples/blinky/$tool

0xc0170, 2015-07-12:
I sent PR which removes -w (workspaces) , did not make any sense to me to split those 2 and then starting having problems like export(name) - is this a project , is this a workspace. It makes more sense that group of projects is a workspace. The only problem for this is how to set workspace output path (where is workspace generated). Ideas?

my idea is like one above:

```
projects:
    group_test:
        k20d50m_blinky: &k20d50m_blinky
            - *module_mbed
            - tools/records/mbed/k20_cmsis.yaml
            - tools/records/mbed/k20_target.yaml
            - tools/records/projects/k20_blinky.yaml
       settings:
         export_dir:
           - path
```

Settings is not much unique, therefore like workspace_settings, sounds better although more inclined to __settings__, which is not yet pgen known :tired_face:

0xc0170, 2015-11-04:
This is now obsolete, I'll propose a new structure for workspaces, to keep them in place but as a nice to have feature, not main.

closed 2015-11-04 by 0xc0170