project-generator/project_generator #199: v0.7

0xc0170, 2015-07-09:
Let's create this for discussion. As I highlighted in doc issue, my suggestions for 0.7

- documentation part of this repo and up to date, including all pgen keywords and commands explained
- testing - basic pgen functionality covered
- internal redesign of some classes , mainly Project() , who holds the data and how to obtain them by tools
- grouping of projects, and workspaces
- import targets from tools (create a new mcu yaml file for pgen definitions)

We already started most of this, so just make tests green with good coverage and all new functionality properly tested.

0xc0170, 2015-07-09:
@matthewelse @sarahmarshy @sg-

0xc0170, 2015-07-09:
Milestone 0.7 - https://github.com/project-generator/project_generator/milestones/v0.7

Add issues which we can track ;)

0xc0170, 2015-07-14:
v0.7 is shaping slowly, I am thinking about merging it this week or at least open PR where we can review all changes which are there (many!)

matthewelse, 2015-07-14:
@0xc0170 @sarahmarshy @sg- I think that we should aim to make sure that v0.7 doesn't drag on for ages like v0.6 - I'm back from holiday and I'm free until I start at ARM on Monday, so I'm happy to work on getting stuff done.

0xc0170, 2015-07-14:
I am doing what I can, fixing things. exporting and building is currently working for me with multiple tools and adding more tests to increase coverage. WHat is currently missing are docs and update valinor (might do some changes to pgen as well).

I would like describe architecture of pgen as it is now, for contributors :-)

sarahmarshy, 2015-07-14:
@matthewelse 

I've got some Jira tickets that should guide us a little bit. Given that the ultimate goal of pgen is to go online, we should progress towards that for a 1.0 release.

https://github.com/project-generator/project_generator/issues/133 
A good start could be this issue. 

Another ticket involves argcomplete, but I took a look at that, and it doesn't seem to be very friendly cross-platform.

We should work on making the init function as clean as possible, as the online tools will need to generate their own yamls.

I created an example program with which to test it for now. You can see it here:
https://github.com/sarahmarshy/pgen_simple_example

We should look at incorporating that example into our new tests.

A quick TODO for init: those lib files you see in that repo are not actually libs and create errors once a uvision project has been generated. We should look at ignoring .lib files when initializing a new project. 

Another side note... since there are now 4 of us contributing, we should try to better document our code internally as well. I know I can be/have been bad about it when contributing to this project, but commenting our code will make all of our lives a little easier.

matthewelse, 2015-07-14:
Yeah I was thinking about the internal documentation, and since I'm planning to work on the general docs, I'll probably have a look at documenting the internals, as well as making sure we have all of the necessary user documentation.

sarahmarshy, 2015-07-14:
Good internal docs should make general docs easier. I will work on commenting some of my portions tomorrow.

sarahmarshy, 2015-07-14:
I think we should get pgen working as we want it before docs. Pgen will potentially change a lot in the coming weeks, and we may need to revise docs multiple times.

matthewelse, 2015-07-14:
Oh that's true good point

matthewelse, 2015-07-14:
So internal docs for now and then external once we have a stable API

0xc0170, 2015-07-26:
Any work in progress? I'll look at testing pgen commands

0xc0170, 2015-07-29:
Besides docs, which will be added later, seems like v0.7 is ready. I'll fix two issues

Test coverage is 79 percent for this release.

closed 2015-07-30 by 0xc0170