project-generator/project_generator #209: Pgen relpaths calculation

sarahmarshy, 2015-07-13:
Currently, pgen uses the function _set_output_dir to calculate the relative ".." identifiers needed to get from a tool project file to the actual project file referenced in the yaml. It adds this "count" to a dictionary that will eventually get passed to tools, and all tools must  call a _fix_paths function to append these ".."s to all filenames. I think this could be simplified using python's os.path.relpath function. 

I think it is in our best interest to simplify these dictionaries that are passed around as much as possible.

You can see how it is currently calculated here:
https://github.com/project-generator/project_generator/blob/dev_v0.7/project_generator/project.py#L391

And how it is used here:
https://github.com/project-generator/project_generator/blob/dev_v0.7/project_generator/tools/uvision.py#L123

You can see an examples of os.path.relpath below:
[image]

0xc0170, 2015-07-13:
Plus to get count of how far we are for root (the number is used in eclipse) and we shold be good to go to change that.

How would this relpath simplify the dictionary ?

sarahmarshy, 2015-07-13:
We would not need count. Could do that in eclipse, like if we have:
```python 
relpath = "..\.."
count = len(relpath.split(os.path.sep))
```
count would then be 2

sarahmarshy, 2015-07-13:
We would also not need that part of the dictionary at all. We could just use the above function in all the tools.

sarahmarshy, 2015-07-13:
We could have a function like this in util.py
```python
def get_relpath(path1, path2):
    return os.path.join(os.path.relpath(path1,path2), path2)
```

Which would go ahead and give the full relative path instead of the shortened version as above.

0xc0170, 2015-07-13:
Let's do that and use this also for workspaces

0xc0170, 2015-07-23:
This hsould be fixed on master, as in project is

```
    @staticmethod
    def _generate_output_dir(path):
        """this is a separate function, so that it can be more easily tested."""

        count = path.count(os.sep) + 1

        return (os.sep.join('..' for _ in range(count)) + os.sep), count
```

closed 2015-07-23 by 0xc0170