project-generator/project_generator #230: Relative paths broken?

sarahmarshy, 2015-07-27:
If I define my export_dir outside of current project directory, IDE cannot find the files needed.

0xc0170, 2015-07-27:
Can you provide an example. What's the current project directory?

Be more specific, so I can reproduce this issue on my side.

sarahmarshy, 2015-07-27:
```python
export_dir:
    -C:/users/sarmar01/Documents/pgen/tests/generated_projects
```

error in uvision:
cmsis_nvic.c: Error:  #5: cannot open source input file "..\..\..\..\..\..\..\..\..\..\mbed\libraries\mbed\targets\cmsis\TARGET_FREESCALE\TARGET_K20XX\TARGET_K20D50M\cmsis_nvic.c": No such file or directory

sarahmarshy, 2015-07-27:
But export dir as normal 
```python
export_dir:
    -generated_projects
```
Works fine

0xc0170, 2015-07-27:
All paths for pgen are relative to the project directory. Pgen does not support absolute paths. we should probably warn a user using absolute paths?

sarahmarshy, 2015-07-27:
Using:
```python
../generated_projects
```

does not seem to work either.

0xc0170, 2015-07-27:
Can you add a test case to tests, should fail and we fix to make it green :+1:

sarahmarshy, 2015-07-27:
Here is fix in pgen_simple 

https://github.com/sarahmarshy/project_generator/commit/ccc61d64a24e29a531c97d65464a4b8cbdf6c997

commit 73ac9786 (2015-07-27): Merge pull request #235 from sarahmarshy/pgen_v0.7

Changes relpath calulation for generated project files. Closes #230
commit c4955b64 (2015-07-27): Changes relpath calulation for generated project files. Closes #230
closed 2015-07-27 by 0xc0170