project-generator/project_generator #271: pgen name - clashes with cython (python parser generator)

0xc0170, 2015-09-02:
We might reconsider using pgen, or at least document that pgen might clash in some cases. 

cython is using pgen (python parser generator), reference: https://github.com/cython/cython/blob/master/setup.py#L124. If project_generator is installed, it would invoke it with a failure.

0xc0170, 2015-09-23:
I proposed few names back in days (reference: https://github.com/project-generator/project_generator/issues/74). We decided pgen which is not a name which should go further. There are some other names we can go with, prgen or projgen .

I'll remove pgen command for v0.8 as it's breaking change, at least project_generator stays.

sg-, 2015-09-24:
what about pg

sg-, 2015-09-24:
or p-g

sg-, 2015-09-24:
or pg-

0xc0170, 2015-09-24:
We need to be careful with really short ones :smile:  That's how we ended up with this issue. 
Here's summary what we are considering:

projgen
proj_gen
p-g
pg-

For me p-g or projgen.

autopulated, 2015-09-29:
progen?
projen?

`progen` is used by a tmake command, but tmake looks to be pretty dead, since Qt migrated to use CMake in ~2008

0xc0170, 2015-10-05:
Thanks James for suggestions, projgen then.. I'll send PR to dev 0.8 branch, to fix this pgen clash.

sg-, 2015-10-05:
progen ftw!

PR #279 (2015-10-05): pgen rename to progen
Should fix #271
closed 2015-10-11 by 0xc0170