project-generator/project_generator #276: Debugger settings do not work correctly in uVision 4

c1728p9, 2015-09-18:
When initially loading project created for uVision the configured debugger does not work.  In the 'debug' tab in options the correct devices is selected yet the load button remains greyed out.  A workaround to fix this locally is to view the 'Utilities' tab and the press 'OK'.  This should be fixed so when a uVision project is opened for the first time everything just works.

sg-, 2015-09-18:
This requires generating another file (uvopt) which store user config. I don't think this is a big deal and certainly not work the entire extra template file at the moment. Its just a side effect of simplicity.

0xc0170, 2015-09-19:
I don't think it requires uvopt to fix this bug. I believe it's just missing some settings. I found out that Utilities were not properly initialized for uvision4.

PR #277 (2015-09-20): Uvision - debugger flash2 settings addition
This should fix #276.

@c1728p9 Please test, seems to be functional now
commit 0f3fba4c (2015-09-20): Uvision - debugger flash2 settings addition

This should fix #276.
closed 2015-09-20 by 0xc0170