project-generator/project_generator #298: Pre/Post build steps not exposed in yaml

sg-, 2015-11-03:
https://github.com/mbedmicro/DAPLink/pull/27

0xc0170, 2015-11-04:
The tools specific options are not exposed in yaml.  It is not trivial to provide options in yaml files for many tools. I'll take uvision as an example. 
It contain hundreds of options. Their internal xml naming, in many cases is not intuitive (does not match the name in the IDE). We used to cover small subset, but this does not scale. Therefore I implemented templates, where we don't need to learn which options what means, we just need to implement parsing - to inject pgen data in, and all the rest is used from a generic/user's template.

To provide options to yaml files, I can think of 2 options.
1. Use templates (open IDE, set, store), for different options, different templates - in most cases I believe there will be one-two templates. This one is implemented.
2. Add back Misc for some tools, this will require to come up with naming (asm, c/c++, linker flags - and their options). Start with very small subset (I don't have enough man days to cover what 1st can cover). This is a request for pre/post, another project might require for uvision to set uARM vs ARM lib, or IAR another option. with this one, we can easily end up close to 1st, but takes much longer path.

I switched to templates for various reason. I am still considering how to enable yaml settings for those tools , but haven't found yet an easy way to implement and for a user to understand. The easiest is open your tool, select what a project requires, store and use.

sg-, 2015-11-04:
I think misc or the equivalent for a given tool and ONLY a given, necessary feature is worthwhile.

0xc0170, 2015-11-07:
#300 deals with this , but no completely. We will provide settings for tools which are documented (for instance uvision command line options). More description is provide in the pull request.

Unfortunately, the option pre/post for uvision is not available via the command line options, could not find it there to be set up, thus for now we are stick with a template (IDE setup).

0xc0170, 2015-11-10:
As I added those flags, I'll add prebuild and postbuild command, should be valid for iar/uvision at least for now

```
iar:
  misc:
    post_build:
      - run_this_script.bat
```

PR #300 (2015-11-07): misc for tools
In progress. Currently covering some basic options for uvision.

```
    uvision:
        misc:
            Cads:
                Optim:
                - 1
                SplitLS:
                - 1
            Aads:
                Ropi:
                - 1
            ArmAdsMisc:
                useUlib:
                - 1
```

This is for advanced users, to get what is ArmAdsMisc and useUlib, requires a lot of work to describe. The easiest way is to check the option how it changes in the project file :-) Thus I was hesitant to cover this, but seems like for advanced proje [... cut]
closed 2016-06-01 by 0xc0170