project-generator/project_generator #339: uvision - debuggers definitions - Flash2 not needed?

0xc0170, 2015-12-29:
We should check debuggers definitions for uvision, what is required, because flash2 there does not make sense and it's same as TargetDlls anyway

ohagendorf, 2016-05-20:
For a functional Keil5 debugger config a 2nd file (*.uvoptx) is necessary incl. a couple of additonal mcu/debugger specific config options.

How should these be added? Within project_generator or as an addition to mbed or ...?

0xc0170, 2016-06-01:
> For a functional Keil5 debugger config a 2nd file (*.uvoptx) is necessary incl. a couple of additonal mcu/debugger specific config options.

Can you elaborate? I noticed uvoptx is generated from .uvproj and should not be required. But had similar view on .ewd file for IAR, and had to add its generation to get debugger configuration.

If it is required, progen should know about it, as it contains debugger settings. As for IAR, specific, generic .ewd is generated. IF needed, ewd can be supplied via templates, and overwrite it.

ohagendorf, 2016-06-02:
Yes, you are right, uvoptx is generated. But unfortunately the project is not usable out of the box. The following happens or have to be done:

- generating project - everything is fine
- compiling - stile everything is fine
- flashing mcu - strange error message, flashing not possible

- it doesn't matter when you do it:
  - open menu Options for Target 'xxx'
  - go to debugger tab
  - press Buttons Settings of choosen debugger
  - close dialog with OK, close Options dialog with OK

- only after this procedure flashing is possible
- just saving (and internally generating uvoptx file) is not enough

Suggestion:
- generating the uvoptx file (some few debugger options are enough) by project_generator
- missing options are added automatically by Keil when saving the project

0xc0170, 2016-06-02:
Thanks for the feedback, so the situation is similar to .ewd IAR file. This is a new feature request, we got only 3 debuggers supported at the moment, should be fairly simple to add this (cmsis-dap,  st-link, jlink) to produce this file. Feel free to send PR, or I'll try to look at it.

However, how come people have not complained yet about debugging. I haven't got problem with flashing so far , for instance in  mbed generated uvision progen projects. I assume there's some specific settings for some MCU which I haven't yet experienced

ohagendorf, 2016-06-02:
I'm using the current version of Keil5, a broad range of STM mcu and exporting projects from mbed. The problem is the same with every mcu and with st-link and cmsis-dap. I didn't test jlink but it wouldn't be a problem to do it.

I'll have a look at it during the next days and try to find a solution.

0xc0170, 2016-06-14:
I run to this problem today, we will need to have a look. I'll create a new issue to track uvopt file generation

0xc0170, 2020-01-03:
> generating the uvoptx file (some few debugger options are enough) by project_generator

This was added (issue 391 above). I believe this is resolved.

closed 2020-01-03 by 0xc0170