project-generator/project_generator #343: 0.8.5 doesnt support empty fields

sg-, 2016-01-07:
Creating template yaml files causes traceback when there is no value or list for a key.

Ex:
```
common:
    group_name:
        - flash_driver
    target:
        - cortex-m0
    includes:
        -
    sources:
        - source
        - source/template
    macros:
        -
```
```
Traceback (most recent call last):
  File "C:\yotta\python\Lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\yotta\python\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "Z:\Development\flashalgo\venv\Scripts\progen.exe\__main__.py", line 9, in <module>
  File "z:\development\flashalgo\venv\lib\site-packages\project_generator\main.py", line 62, in main
    return args.func(args)
  File "z:\development\flashalgo\venv\lib\site-packages\project_generator\commands\generate.py", line 29, in run
    if project.generate(args.tool, copied=args.copy, copy=args.copy) == -1:
  File "z:\development\flashalgo\venv\lib\site-packages\project_generator\project.py", line 549, in generate
    files = exporter(self.project['export'], self.settings).export_project()
  File "z:\development\flashalgo\venv\lib\site-packages\project_generator\tools\uvision.py", line 382, in export_project
    path, files = self._export_single_project('uvision') #todo: uvision will switch to uv4
  File "z:\development\flashalgo\venv\lib\site-packages\project_generator\tools\uvision.py", line 347, in _export_single_project
    uvproj_dic['Project']['Targets']['Target']['TargetOption']['TargetArmAds'], expanded_dic)
  File "z:\development\flashalgo\venv\lib\site-packages\project_generator\tools\uvision.py", line 202, in _uvproj_set_TargetArmAds
    uvproj_dic['Cads']['VariousControls']['Define'] = ', '.join(project_dic['macros']).encode('utf-8')
TypeError: sequence item 0: expected string, NoneType found
```

0xc0170, 2016-01-07:
Is this something we want to support , having empty values in the yaml, is it considered correct?

```
    includes:
        -
```

I can add catching those errors, will look at this and write some tests

sg-, 2016-01-07:
It works for some but not others which is inconsistent behavior (worse case of all!!)

I like the idea of empty fields so you don't have to read the docs for simple modifications. Adding commonly used placeholders is a +1 IMO and a regression from the 0.6x series

0xc0170, 2016-01-07:
> Adding commonly used placeholders is a +1 

Can you elaborate?

I'll look at this uvision error, its connected to encoding.

PR #344 (2016-01-08): Project - eliminate None - empty values from YAML
should fix #343 

I'll add tests, will test regression first
closed 2016-01-08 by 0xc0170