project-generator/project_generator #361: -b build order for uvision

sg-, 2016-03-10:
Not sure what order this builds in but it doesn't seem to match the projects.yaml file or be alphabetically.

I think it should build according to projects.yaml entry order.

For example: https://github.com/mbedmicro/DAPLink/blob/master/projects.yaml yeilds:
```
(venv) C:\Developmnent\daplink> progen generate -t uvision -b
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmkl02z_if\k20dx_frdmkl02z_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_sscity_if\lpc11u35_sscity_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\sam3u2c_bl\sam3u2c_bl.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmk20dx_if\k20dx_frdmk20dx_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_arm_watch_nrf51_if\lpc11u35_arm_watch_nrf51_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmkw24f_if\k20dx_frdmkw24f_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_rblnano_if\k20dx_rblnano_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_arm_watch_stm32f411_if\lpc11u35_arm_watch_stm32f411_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\sam3u2c_nrf51dk_if\sam3u2c_nrf51dk_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmkl46z_if\k20dx_frdmkl46z_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_lpc824xpresso_if\lpc11u35_lpc824xpresso_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\sam3u2c_nrf51dongle_if\sam3u2c_nrf51dongle_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\kl26z_microbit_if\kl26z_microbit_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_bl\k20dx_bl.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_hrm1017_if\lpc11u35_hrm1017_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_archmax_if\lpc11u35_archmax_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmkl05z_if\k20dx_frdmkl05z_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_ssci824_if\lpc11u35_ssci824_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\kl26z_bl\kl26z_bl.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmk22f_if\k20dx_frdmk22f_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_c027_if\lpc11u35_c027_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_arm_watch_efm32_if\lpc11u35_arm_watch_efm32_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_archpro_if\lpc11u35_archpro_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\sam3u2c_nrf51mkit_if\sam3u2c_nrf51mkit_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_ssci1114_if\lpc11u35_ssci1114_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_rbl_if\k20dx_rbl_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_tiny_if\lpc11u35_tiny_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_archlink_if\lpc11u35_archlink_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_lpc812xpresso_if\lpc11u35_lpc812xpresso_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmkl25z_if\k20dx_frdmkl25z_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmkl26z_if\k20dx_frdmkl26z_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_efm32gg_stk_if\lpc11u35_efm32gg_stk_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\lpc11u35_archble_if\lpc11u35_archble_if.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\k20dx_frdmk64f_if\k20dx_frdmk64f_if.uvproj
```

0xc0170, 2016-03-11:
Thanks for reporting, will have a look at this

0xc0170, 2016-03-13:
Looking at pyYAML if it supports orders, the answer is (reference: http://pyyaml.org/ticket/29): "The spec clearly indicates that the key order is a representation detail and should not be used for constructing native objects.". There are workarounds to get the order though.

I don't think this is specific to build, it's for any command, which uses generator. That one loads all the project.

The easiest would be order them alphabetically. Or the harder way, to change yaml loader, might not be safe

PR #400 (2016-06-17): Sort projects by name in all commands
resolves #361
commit e1c5161e (2016-06-17): Sort projects by name in all commands

resolves #361
closed 2016-06-17 by 0xc0170