project-generator/project_generator #364: IAR - Need to add a template for .ewd file

RyoheiHagimoto, 2016-03-16:
The debugger setting file is .ewd for the IAR IDE.
The .ewd file is not a template, but generated by a project generator regardless target and no way to configured.
The generated .ewd file is for cortex-M, and is not suitable for GR-PEACH (Cortex-A9) debug setting.
Therefore, users of GR-PEACH have to modify CMSIS-DAP section for debugger manually.
To configure .ewd file, please add a template for .ewd file.

We want to modify following:
- value of "state" in "CMSISDAPResetList"
- value of "state" in "CMSISDAPInterfaceRadio"

The former specifies the reset method. The value "10" is not valid to cortex-A9. Threfore IDE choose the default setting.
The latter specifies the interface is whether JTAG or SWD. I think that some platforms need to modify this setting.

0xc0170, 2016-03-16:
Thanks for reporting I noticed this on mbed.org. I'll have a look how to provide this to users. We definitely should add support for .ewd files. 

> The former specifies the reset method. The value "10" is not valid to cortex-A9. Threfore IDE choose the default setting.

I can't recall what 10 value defines for cortex-m.

> The latter specifies the interface is whether JTAG or SWD. I think that some platforms need to modify this setting.

https://github.com/project-generator/project_generator_definitions/blob/master/project_generator_definitions/target/targets.py#L360, This should set jtag or swd. Can you please test it , to add interface to your target (look at the target above how it defines interface) and see if this value is generated? We might fix this easily as it's generic option.

0xc0170, 2016-03-16:
@RyoheiHagimoto I am following the https://developer.mbed.org/teams/Renesas/wiki/Exporting-to-IAR-Systems, and the interface for me is grayed out in IAR (using 7.10 version). JTAG is selected. Any pointers why?

RyoheiHagimoto, 2016-03-17:
@0xc0170 
> >The former specifies the reset method. The value "10" is not valid to cortex-A9. Threfore IDE choose the default setting.

> I can't recall what 10 value defines for cortex-m.

I guess 10 value defines "Reset and halt bootloader (default)".

>Can you please test it , to add interface to your target (look at the target above how it defines interface) and see if this value is generated? We might fix this easily as it's generic option.

I tried, but interface setting was not changed.
I added following:

> 

        'debugger': {
            'name': 'cmsis dap',
            'interface': 'swd',
        }

I think 'name' I set is not correct.
I could not find 'name' string that maches this case.
Please tell me what string is correct in this case.

> I am following the https://developer.mbed.org/teams/Renesas/wiki/Exporting-to-IAR-Systems, and the interface for me is grayed out in IAR (using 7.10 version). JTAG is selected. Any pointers why?

I guess it is caused by version of tool.
I confirmed version 7.40.3 and version 7.50.3.

0xc0170, 2016-03-17:
``cmsis-dap`` - this is the name. I'll have 7.50.x on my another PC, I'll test it tonight if I can get swd/jtag not grayed-out. Please test if the interface is properly set with the change above.

RyoheiHagimoto, 2016-03-18:
@0xc0170 Thank you for your support. I can get SWD is selected in IAR IDE.

RyoheiHagimoto, 2016-04-07:
@0xc0170 I am looking forward to addition of the template for .ewd file. Could you please tell me the plan of it?

0xc0170, 2016-04-07:
It should be the first thing for progen in the upcoming days. I am having problems with the flash debugger settings for some devices, I assume it's connected to ewd as well :/

0xc0170, 2016-04-07:
I sent PR adding .ewd support :-) please review it

closed 2016-05-07 by 0xc0170

RyoheiHagimoto, 2016-05-18:
@0xc0170 From [export directory of mbed-SDK](https://github.com/mbedmicro/mbed/tree/master/workspace_tools/export), mbed-sdk supports 2 platforms.
I think that once iar.py is modified for template for ewd file, each platform for IAR need ewd.tmpl file.
I can add iar_rz_a1h.ewd.tmpl, but I can NOT add iar_nucleo_f746zg.ewd.tmpl, because I do not maintain nucleo_f746zg. Could you please modify mbed-SDK to support ewd file?
I will modify iar_rz_a1h.ewd.tmpl after your modification.

PR #375 (2016-05-07): IAR - .ewd file templates
It should fix #364. Adding multimple templates to the picture. This relies on the
extension of the templates. Previously, only one template was supported, thus we did
not care about extension.

@RyoheiHagimoto  Can you please test ? Add to template list for progen, .ewd file. I have to add more sugar to accept .ewd.tmpl as there are in mbed. I quickly tested .ewp and .ewd template files (removed .tmpl extension).

```
ewd file - .ewd
ewp file - .ewp

change in iar in exporters in mbed, the template names to .ewd and .ewl , same for files there, should work
```
commit 1247fbae (2016-05-07): IAR - .ewd file templates

It should fix #364. Adding multimple templates to the picture. This relies on the
extension of the templates. Previously, only one template was supported, thus we did
not care about extension.