project-generator/project_generator #385: Define flags are wrong for armasm

c1728p9, 2016-05-31:
The command line flags which define values for ARMASM are wrong.  Currently it generates with '-d'.  It needs to switch to '--pd'.

0xc0170, 2016-05-31:
This is for predefined: http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0489h/BGBDJFAA.html

> --predefine is not equivalent to the compiler option -Dname. --predefine defines a global variable whereas -Dname defines a macro that the C preprocessor will expand.
Although you can use predefined global variables in combination with assembly control directives, for example IF and ELSE to control conditional assembly, they are not intended to provide the same functionality as the C preprocessor in the assembler. If you need this functionality, ARM recommends you use the compiler to pre-process your assembly code.

By default in progen, uvision is set to preprocess assembly files, it should work.

0xc0170, 2016-05-31:
I think to resolve this, the assembly files should use preprocesor and proper defines rather predefines. Look at the syntax how to: http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0489h/BGBDJFAA.html

```
// in the assembly file

#ifdef THIS_FEATURE
  some asm code
#else
  another asm code 
#endif
```

closed 2016-05-31 by 0xc0170