project-generator/project_generator #388: uvision build is to strict :)

sg-, 2016-06-01:
SUCCESS_VALUE should be 0 or 1 for uvision (warnings are OK in some cases)

https://github.com/project-generator/project_generator/blob/0a15760ab50fee176f95a2db1c027a7a01b84ded/project_generator/tools/uvision.py#L110

```
progen.tools.uvision ERROR      Project: warnings build failed with the status: projectfiles\uvision\lpc1114fn28\lpc1114fn28.uvproj
progen.tools.uvision ERROR      Project: errors build failed with the status: projectfiles\uvision\mk80f25615\mk80f25615.uvproj
progen.tools.uvision INFO       Project: success (0 warnings, 0 errors) build succeeded with the status: projectfiles\uvision\efm32gg\efm32gg.uvproj
```

0xc0170, 2016-06-02:
I'll fix it, 0 and 1 should be fine.

closed 2016-06-02 by 0xc0170

sg-, 2016-06-02:
Let me know when a version is published so I can update the version used to get this fix

0xc0170, 2016-06-02:
It was published, its out v0.9.2

PR #389 (2016-06-02): Error message revision and uvision build warnings
Does not fail silently for invalid project name. Corrected message order for successful uvision build. Closes #388
commit b613dbbf (2016-06-02): Does not fail silently for invalid project name. Corrected message order for successful uvision build. Closes #388