project-generator/project_generator #403: Duplicate C source files in generated project

LMESTM, 2016-06-22:
Just want to share a problem I met yesterday: 
I used master branch (Merge pull request #399 from project-generator/dedupe-2-fn ) and the resulting project (a mbed project generated into a µvision project) contained a lot of duplicate source files and which resulted in build errors. I want back to a know working version 0.8.16 and this is working fine again.  The version I tested is not yet a released version, so not sure if this is relevant or not.

0xc0170, 2016-06-22:
It is definitely good to share, I am currently testing the master branch, not yet released the new version.

cc @theotherjimmy

theotherjimmy, 2016-06-22:
It seems that the duplication is only in the .uvproj file, not in the source tree. I'm looking into why now.

theotherjimmy, 2016-06-22:
That was silly of me. I think I squashed this bug. Thanks for reporting!

theotherjimmy, 2016-06-22:
@LMESTM Could you test again?

LMESTM, 2016-06-23:
I confirm there is no more duplication in .uvproj file so the project builds fine now, thanks for that !
.. but (sorry there is a minor but ;-) ) there are now empty directories visible in the project in µvision5. I think those  are directories where we can only find header files like hal/hal and hal/api in mbed. 
[image]

0xc0170, 2016-06-23:
@LMESTM When you report a bug, please provide as much details as possible to reproduce it. What commands are you using ? to build and then to export, I just tried KL25Z MBED_10 project, I noticed another issues, but not empty directories. using the latest master on mbed and progen.

0xc0170, 2016-06-23:
Confirmed, I just rebased my progen branch, I can see those empty directories. We will fix it

0xc0170, 2016-06-24:
@LMESTM Should be resolved

closed 2016-06-24 by 0xc0170

LMESTM, 2016-06-24:
@0xc0170  Indeed it works fine now - THX

PR #406 (2016-06-24): Iterate - group att missing, groups with headers were empty
Should fix #403.

cc @theotherjimmy
commit d820f365 (2016-06-24): Iterate - group att missing, groups with headers were empty

Should fix #403