project-generator/project_generator #407: TypeError: a bytes-like object is required, not `_io.TextIOWrapper`

jamesadevine, 2016-06-29:
https://github.com/project-generator/project_generator/blob/master/project_generator/tools/uvision.py#L304

The above line fails when using python 3.

I've applied this fix to my installation:

```python
uvproj_dic = xmltodict.parse(open(template, encoding="utf8").read())
```
And it works!

I would form this into a PR, but I currently don't have the time!

theotherjimmy, 2016-06-29:
I have your back on this one. I'll make the PR for you.

theotherjimmy, 2016-06-29:
I'll even give you the credit, you did the heavy lifting on this one.

0xc0170, 2016-06-29:
Thanks for reporting and providing a fix !

closed 2016-06-29 by 0xc0170

jamesadevine, 2016-06-30:
Thanks both! :smile:

jamesadevine, 2016-06-30:
@theotherjimmy @0xc0170 I failed to realise that the "encoding" parameter fails on < python 3! 😱

0xc0170, 2016-06-30:
> @theotherjimmy @0xc0170 I failed to realise that the "encoding" parameter fails on < python 3! :scream:

Ehm, let me check the tests if that condition is not tested. I bet it's not as we have to run some tests on the real examples with real project templates. I have been exporting prior the release and did not have any failure.

0xc0170, 2016-06-30:
Anyway, the patch is wrong , this condition is not covered - https://github.com/theotherjimmy/project_generator/blob/4712c1d64d6cb2293ce16b6a044c767a26e8e420/project_generator/tools/uvision.py#L307

Thus I might have not spotted it while running local tests..

PR #408 (2016-06-29): Fix python 3 type error
Resolves #407
commit 4712c1d6 (2016-06-29): Fix python 3 type error

Resolves #407