project-generator/project_generator #419: Relative paths in progen

sarahmarshy, 2016-07-20:
If I pass a project dict to a progen project, with the paths already set to be relative to the export directory, there are issues. 

This function:
```python
def _generate_output_dir(settings, path):
        """ This is a separate function, so that it can be more easily tested """
        relpath = os.path.relpath(settings.root,path)
        count = relpath.count(os.sep) + 1

        return relpath+os.path.sep, count
```
used in this context:
````python
self.project['export']['output_dir']['rel_path'], self.project['export']['output_dir']['rel_count'] = self._generate_output_dir(self.settings, path)
```
Where rel_path is FINALLY used:
```python
 fix_paths(self.project['export'], self.project['export']['output_dir']['rel_path'], list(FILES_EXTENSIONS.keys()) + ['include_paths', 'source_paths'])
```
Where fix_paths is defined as:
```python
def fix_paths(project_data, rel_path, extensions):
    """ Fix paths for extension list """
    norm_func = lambda path : os.path.normpath(os.path.join(rel_path, path))
    for key in extensions:
        if type(project_data[key]) is dict:
            for k,v in project_data[key].items():
                project_data[key][k] = [norm_func(i) for i in v]
        elif type(project_data[key]) is list:
            project_data[key] = [norm_func(i) for i in project_data[key]]
        else:
            project_data[key] = norm_func(project_data[key])
```

Is there a specific reason for this complexity? Why can't we just use the built in method relpath to make all files relative to the export directory. There should also be some functionality when paths are ALREADY relative to the export directory. It just seems crazy to MANUALLY append relative prefixes(..\\..\\....) to every path. Should relative paths be disallowed in the yaml/project dicts then?

sarahmarshy, 2016-07-26:
This is also dependent on self.settings.root, which is set to be the cwd if one is not provided. If there is no common root, we should think of a better way to handle it than cwd.

0xc0170, 2016-07-26:
> If I pass a project dict to a progen project, with the paths already set to be relative to the export directory, there are issues. 

Can you write a test case for that to show it fails?

> Is there a specific reason for this complexity? 

The reason there is that yaml files reference files from a root, and an exported dir might not be in the root, thus tweaking paths is required. 

> Why can't we just use the built in method relpath to make all files relative to the export directory.

Isn't that what function fix_paths does? Loops through its data and set a correct path? Any suggestions for improvements?

> There should also be some functionality when paths are already relative to the export directory. 

This is not a common use case, and currently disallowed. how would you provide this functionality?

> we should think of a better way to handle it than cwd. 

suggestions welcome

sarahmarshy, 2016-07-26:
The complexity I am talking about is manually appending relative prefixes to all filepaths. Why not use python's built in relpath?

sarahmarshy, 2016-07-26:
>Can you write a test case for that to show it fails?

>This is not a common use case, and currently disallowed. how would you provide this functionality?

If it is disallowed, we know what the test will show, right?

0xc0170, 2016-07-26:
> The complexity I am talking about is manually appending relative prefixes to all filepaths. That is what fix_paths does. Why not use python's built in relpath?

Send a patch please to fix it

0xc0170, 2020-01-04:
I'll close this one, has been some time.

closed 2020-01-04 by 0xc0170