project-generator/project_generator #422: IAR debug broken

sarahmarshy, 2016-07-25:
IAR debug does not work. I have found a solution: turning on the use of flash loaders (set in .ewd file). Is there a reason this was not enabled?

0xc0170, 2016-07-26:
What boards did you test? Various MCU vendors? like nxp, emf32, atmel etc?

I know that for instance nxp (freescale) boards require use flash loader to be enabled, but some others don't. Based on that, I assumed that this is a target specific, and we enabled .ewd templates.

0xc0170, 2016-07-26:
Thanks for raising this, this has been a known issue (I would like to understand those flashloaders better) that was not tracked.

0xc0170, 2016-08-08:
Any update to this issue?

0xc0170, 2019-12-06:
If yet not fixed, reopen via new issue.

closed 2019-12-06 by 0xc0170