project-generator/project_generator #425: Extend exporter API to return source of project files instead of writing them to disk

theotherjimmy, 2016-07-26:
The exporters in mbed are undergoing a huge revision and part of that revision is a zero-copy strategy. The goal is to create a zip file from an mbed project, using progen, and enforce that only the final zip file is written to disk. Currently, the only files that have two copies (one in the zip, one in the build directory) are:
 - `mbed_config.h` (not progen's problem)
 - project files, generated by progen

The feature request is this: I would like an API from the exporters that allows me to generate the project files without generating them to a file.

sarahmarshy, 2016-07-27:
https://github.com/sarahmarshy/project_generator/tree/return_raw_xml

sarahmarshy, 2016-07-27:
I made a branch [here](https://github.com/sarahmarshy/project_generator/commits/raw_xml)

0xc0170, 2016-07-28:
What would be the function name? Exporter API provides ``export_project`` that should export a project file. This could be ``get_export_projects()`` or ``get_export_dicts()`` ?

Be aware of limitation for exporters that use jinja2 templates (they use ``gen_file_jinja``). As a template contains the magic. A user would need to get the export_dicts, and use ``gen_file_jinja`` or do something similar. 

We might want to signal somehow what exporter uses (raw vs jinja2), a new method within Tools exporters?

theotherjimmy, 2016-07-28:
>Be aware of limitation for exporters that use jinja2 templates (they use gen_file_jinja). As a template contains the magic. A user would need to get the export_dicts, and use gen_file_jinja or do something similar. 

I don't think that's a limitation. We would use something like https://github.com/project-generator/project_generator/blob/master/project_generator/tools/tool.py#L151and instead of doing the next line, we would write it to a dict. afterwards, we could just return said dict instead of what the generate normally returns.

sarahmarshy, 2016-07-28:
New branch:
https://github.com/sarahmarshy/project_generator/tree/raw_proj_files

0xc0170, 2020-01-04:
> https://github.com/sarahmarshy/project_generator/tree/raw_proj_files

I'll take this branch and create PR, will review it there and see what we can do with this one. I'll close this in the meantime.

closed 2020-01-04 by 0xc0170