project-generator/project_generator #428: IAR dictionary vs jinja template

sarahmarshy, 2016-08-01:
Why did we move from IAR templates to the dictionary? I found a solution to the FPU problem, but editing the XML generation for IAR is a nightmare right now because that dictionary is unreadable. Can we return to the template?

@0xc0170  @theotherjimmy

0xc0170, 2016-08-08:
> I found a solution to the FPU problem

Is there any FPU problem? Are you referring to the IAR syntax change (FPU vs FPU2)? I dont recall there's any issue opened here, please create one.

> Why did we move from IAR templates to the dictionary?

It should be easier to maintain, keep everything within python. 
To maintain the dict + jinja2 template vs one dict.  To fill in jinja2 (the initial port work) is much more time consuming (every single option/checkbox should be allowed to be changed, thus a template would be a lot of work). I found it much easier to work only with python dic that maps to 1:1 to the provided tool syntax.

> is a nightmare right now because that dictionary is unreadable

jinja2 would be also :-)

sarahmarshy, 2016-08-08:
They match 1:1, but the embedding the XML syntax into a dictionary is convoluted, because the tags have no logical significance within progen. Though they are obviously significant _in IAR_. Take this line: https://github.com/project-generator/project_generator/blob/master/project_generator/tools/iar.py#L140. These keys ewp_dic[index_iccarm]['data']['option'], that prefix everything do nothing to make each value unique for me. This is the important key: 'CCDefines'. I think it is easier to work with that key only, as you would in a template. The dictionary as it is, it is difficult to read the nested structure. If there are many elements within one parent, it is hard to distinguish the parent and parent's parent.

theotherjimmy, 2016-08-08:
I'm going to provide another perspective:
It looks awful. This line https://github.com/project-generator/project_generator/blob/master/project_generator/tools/iar.py#L40 is like 10,000 characters long! That's just impossible to read, and much worse to edit. We could at least store the data as an XML blob and read it in, just like the template (https://github.com/project-generator/project_generator/blob/master/project_generator/tools/iar.py#L386)

I'm not sure we even need jinja2 to get this right, but we need to get rid of those incomprehensible ordered dictionaries.

theotherjimmy, 2016-08-08:
> I found it much easier to work only with python dic that maps to 1:1 to the provided tool syntax. for instance: https://github.com/project-generator/project_generator/blob/master/project_generator/tools/iar.py#L149 , get the dict, fill in progen data, write.

I find that code superfluously complicated, and fragmented.

bridadan, 2016-08-08:
Whether or not Jinja2 is the right tool could be argued. But we should strive to keep this type of "data" out of the logic.

0xc0170, 2016-08-08:
> I'm not sure we even need jinja2 to get this right, but we need to get rid of those incomprehensible ordered dictionaries.

Sounds good to have it as a template and read it. I personally haven't profiled it how much time it would take. Let's do that and we can start from there. I am not against having it stored as a file (real project file), load it, edit and generate.

theotherjimmy, 2016-08-08:
I really don't think that reading in 5KB of XML vs 3KB of Python is going to be that big of a difference. I'll profile for you anyway.

0xc0170, 2016-08-09:
Any progress ?

theotherjimmy, 2016-08-09:
See #435

0xc0170, 2016-08-10:
merged, the templates are now valid project files we use. I'll close this one.

closed 2016-08-10 by 0xc0170