project-generator/project_generator #429: [RC5 OoB] armasm errors for --preinclude option

toyowata, 2016-08-03:
How to reproduce:
```
mbed import https://github.com/ARMmbed/mbed-os-example-ble
mbed update mbed-os-5.1.0-rc5
cd BLE_HeartRate
# Add K22F config in the mbed_app.json
mbed export -i uvision -m K22F
```

I got errors below for the armasm.

```
*** Using Compiler 'V5.06 update 2 (build 183)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'Unnamed_Project'
assembling startup_MK22F51212.S...
startup_MK22F51212.S: error: A3900U: Unrecognized option '--preinclude=mbed_config.h'.
assembling SVC_Table.S...
SVC_Table.S: error: A3900U: Unrecognized option '--preinclude=mbed_config.h'.
".\.build\uvision4\Unnamed_Project.axf" - 2 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:02
```
cc @sg- @theotherjimmy

sarahmarshy, 2016-08-03:
Hi Toyo, see my comment on #2347. I found the commit that is responsible.

toyowata, 2016-08-03:
@sarahmarshy I don't think the armasm support --preinclude option.  Why do you think this is an mbed issue?

sarahmarshy, 2016-08-03:
@toyowata the line that I have just changed with this [commit](https://github.com/ARMmbed/mbed-os/pull/2347/commits/4484e2e14bb0dc207aeef4cf92558fd3fd7d60ac) was adding the --preinclude option to the assembler flags.

0xc0170, 2016-08-08:
@toyowata Sarah is correct. Progen does not know and never will what is a valid option for tools (uvision uses proprietary xml for instance). therefore a consumer of progen (in this in this case mbed) needs to take care of how to make preinclude work with assembly files.

I noticed this was fixed in the mbed, not to use preinclude. I'll close this issue

closed 2016-08-08 by 0xc0170