project-generator/project_generator #430: [RC5 OoB] armasm errors

samchuarm, 2016-08-03:
steps to reproduce:

```
mbed import https://github.com/ARMmbed/mbed-os-example-client
mbed update mbed-os-5.1.0-rc5
<Add WIFI config in mbed_app,json>
mbed export -i uvision -m NUCLEO_F401RE
```

I got errors as below:

```
*** Using Compiler 'V5.06 update 2 (build 183)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'Unnamed_Project'
.............
compiling cmsis_nvic.c...
assembling startup_stm32f401xe.S...
startup_stm32f401xe.S: error: A3900U: Unrecognized option '--preinclude=mbed_config.h'.
.............
assembling SVC_Table.S...
SVC_Table.S: error: A3900U: Unrecognized option '--preinclude=mbed_config.h'.
.............
".\.build\uvision4\Unnamed_Project.axf" - 2 Error(s), 37 Warning(s).
Target not created.
```

sarahmarshy, 2016-08-03:
Duplicated here: https://github.com/project-generator/project_generator/issues/429

0xc0170, 2016-08-08:
Please use search functionality, not to create duplicates. Thanks

closed 2016-08-08 by 0xc0170