project-generator/project_generator #431: [RC5 OOB] error of mbed export to make GCC_ARM

samchuarm, 2016-08-03:
Steps to reproduce:

```
mbed import https://github.com/ARMmbed/mbed-os-example-client
mbed update mbed-os-5.1.0-rc5
mbed compile -m RZ_A1H -t GCC_ARM -c
mbed export -m RZ_A1H -i GCC_ARM
cd projectfiles\gcc_arm_RZ_A1H
make
```

I got the errors as below:

```
..\..\../mbed-os/hal/common/mbed_board.c: In function 'mbed_error_vfprintf':
..\..\../mbed-os/hal/common/mbed_board.c:85:9: error: 'for' loop initial declara
tions are only allowed in C99 or C11 mode
         for (int i = 0; i < size; i++) {
         ^
..\..\../mbed-os/hal/common/mbed_board.c:85:9: note: use option -std=c99, -std=g
nu99, -std=c11 or -std=gnu11 to compile your code
make[1]: *** [mbed-os/hal/common/mbed_board.o] Error 1
make: *** [all] Error 2

```

sarahmarshy, 2016-08-03:
@theotherjimmy

theotherjimmy, 2016-08-03:
mbed export to gcc_arm is not handled by progen.

This template is bad: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/gcc_arm_rz_a1h.tmpl
It uses constant CC_FLAGS, which will never update correctly from the toolchain c_flags.

0xc0170, 2016-08-08:
@theotherjimmy / @samchuarm  Please create an issue on mbed for this. I'll close this issue as it's not relevant to this module

closed 2016-08-08 by 0xc0170