project-generator/project_generator #434: [RC5 OOB] compile error when export to Keil for K64F

sunsmilearm, 2016-08-04:
Commands I used:
mbed import mbed-os-example-client
mbed update mbed-os-5.1.0-rc5
mbed export -i uvision -mK64F

Then open the project in Keil5, get compile errors:

assembling startup_MK64F12.S...
startup_MK64F12.S: error: A3907U: Via file '.\.build\uvision4\startup_mk64f12._ia' command too long for buffer.
assembling SVC_Table.S...
SVC_Table.S: error: A3907U: Via file '.\.build\uvision4\svc_table._ia' command too long for buffer.

sarahmarshy, 2016-08-04:
Duplicated here: https://github.com/project-generator/project_generator/issues/429

0xc0170, 2016-08-08:
@sunsmilearm I have seen this error, there are limits (preprocessor has a buffer). mbed uses a lot of defines, that triggers this error. Plus there's a windows limitation.

@sarahmarshy Is this a duplicate? The issue referenced above is about preinclude, this one is about too long buffer.

We will hit this error often. I don't think this is related to progen, and might be better to track this issue under mbed as "how to reduce the -D length for uvision".

sarahmarshy, 2016-08-08:
@0xc0170 solved with https://github.com/ARMmbed/mbed-os/pull/2377

sarahmarshy, 2016-08-08:
You are right that this is not a duplication of the link above. My mistake. However, there have been a few duplicated issues about this in progen.

0xc0170, 2016-08-08:
> You are right that this is not a duplication of the link above. My mistake. However, there have been a few duplicated issues about this in progen.

Yes, users should use search functionality prior creating a new one.

> @0xc0170 solved with ARMmbed/mbed-os#2377

Thanks. I'll close this one. however, as noted above, we might hit this limit again (there are limits in windows and also in tools (they have buffers and how they interact (asm with preprocessor, etc)).

closed 2016-08-08 by 0xc0170